espressif/esp-now #1: Platformio  build error: 'CONFIG_ESPNOW_CHANNEL' undeclared

NoobSmall, 2021-09-10:
Where can I correct this error?

Building in release mode
Compiling .pio\build\esp32dev\src\main.o
src/main.c: In function 'example_espnow_task':
src/main.c:225:41: error: 'CONFIG_ESPNOW_CHANNEL' undeclared (first use in this function); did you mean 'CONFIG_SPIFFS_CACHE'?
                         peer->channel = CONFIG_ESPNOW_CHANNEL;
                                         ^~~~~~~~~~~~~~~~~~~~~
                                         CONFIG_SPIFFS_CACHE
src/main.c:225:41: note: each undeclared identifier is reported only once for each function it appears in
src/main.c:228:43: error: 'CONFIG_ESPNOW_LMK' undeclared (first use in this function); did you mean 'CONFIG_ESP_INT_WDT'?
                         memcpy(peer->lmk, CONFIG_ESPNOW_LMK, ESP_NOW_KEY_LEN);
                                           ^~~~~~~~~~~~~~~~~
                                           CONFIG_ESP_INT_WDT
In file included from C:/Users/L_D/.platformio/packages/framework-espidf/components/esp_timer/include/esp_timer.h:44,
                 from C:/Users/L_D/.platformio/packages/framework-espidf/components/freertos/xtensa/include/freertos/portmacro.h:84,
                 from C:/Users/L_D/.platformio/packages/framework-espidf/components/freertos/include/freertos/portable.h:95,
                 from C:/Users/L_D/.platformio/packages/framework-espidf/components/freertos/include/freertos/FreeRTOS.h:107,
                 from src/main.c:19:
src/main.c: In function 'example_espnow_init':
src/main.c:301:49: error: 'CONFIG_ESPNOW_PMK' undeclared (first use in this function); did you mean 'CONFIG_ESP_INT_WDT'?
     ESP_ERROR_CHECK( esp_now_set_pmk((uint8_t *)CONFIG_ESPNOW_PMK) );
                                                 ^~~~~~~~~~~~~~~~~
C:/Users/L_D/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
src/main.c:312:21: error: 'CONFIG_ESPNOW_CHANNEL' undeclared (first use in this function); did you mean 'CONFIG_SPIFFS_CACHE'?
     peer->channel = CONFIG_ESPNOW_CHANNEL;
                     ^~~~~~~~~~~~~~~~~~~~~
                     CONFIG_SPIFFS_CACHE
src/main.c:332:25: error: 'CONFIG_ESPNOW_SEND_COUNT' undeclared (first use in this function); did you mean 'CONFIG_ESP32_REV_MIN'?
     send_param->count = CONFIG_ESPNOW_SEND_COUNT;
                         ^~~~~~~~~~~~~~~~~~~~~~~~
                         CONFIG_ESP32_REV_MIN
src/main.c:333:25: error: 'CONFIG_ESPNOW_SEND_DELAY' undeclared (first use in this function); did you mean 'CONFIG_ESP32_REV_MIN_0'?
     send_param->delay = CONFIG_ESPNOW_SEND_DELAY;
                         ^~~~~~~~~~~~~~~~~~~~~~~~
                         CONFIG_ESP32_REV_MIN_0
src/main.c:334:23: error: 'CONFIG_ESPNOW_SEND_LEN' undeclared (first use in this function); did you mean 'CONFIG_ESP32_REV_MIN'?
     send_param->len = CONFIG_ESPNOW_SEND_LEN;
                       ^~~~~~~~~~~~~~~~~~~~~~
                       CONFIG_ESP32_REV_MIN

zhanzhaocheng, 2021-09-23:
You need to add `espnow.h` to your `main.c`

NoobSmall, 2021-09-24:
> 您需要添加`espnow.h`到您的`main.c`
I have already added it, but it still doesn't work

NoobSmall, 2021-09-24:
> You need to add `espnow.h` to your `main.c`
I have already added it, but it still doesn't work。
Add the definition in the config file and compile successfully

lhespress, 2023-01-11:
@NoobSmall You can reference the solution example which use all components of the project.

closed 2023-04-14 by lhespress