espressif/esp-now #2: I already have the DEVELOPMENT environment for ESP-IDF. How can I integrate ESP-NOW into the IDF project

lybgo, 2021-09-23:
I already have the DEVELOPMENT environment for ESP-IDF. How can I integrate ESP-NOW into the IDF project

zhanzhaocheng, 2021-09-23:
Add `set(ESPNOW_PATH /path/to/esp-now/)` in `CMakeLists.txt` in your project directory
or 
Add environment variable `export ESPNOW_PATH=/path/to/esp-now/`

lybgo, 2021-09-23:
I have set ```set(ESPNOW_PATH /path/to/esp-now/)```, but there is an error
[image]
.
[image]

lybgo, 2021-09-24:
Need to add more code: ```set(EXTRA_COMPONENT_DIRS ${ESPNOW_PATH}/components)```

closed 2021-09-24 by lybgo