espressif/esp-now #3: Bind by RSSI example

poky, 2021-09-25:
Hi, thank you for the library!
In the promote video shows that the bind is set to within the 4cm RSSI range, could you show where I can find the example of it? 
I'm testing the control example with 3 C3 dev boards, but doesn't show this feature.

Thanks!

[image]

zhanzhaocheng, 2021-09-26:
Currently, the RSSI between the two devices is used as the filter condition for the distance. 

You can modify the filtered data frame through the following configuration (`frame_head.filter_weak_signal`)

```c

                espnow_frame_head_t frame_head = {
                    .retransmit_count = 15,
                    .broadcast        = true,
                    .channel          = ESPNOW_CHANNEL_ALL,
                    .filter_weak_signal = -20,
                };

                espnow_ctrl_data_t data = {
                    .initiator_attribute = ESPNOW_ATTRIBUTE_KEY_1,
                    .responder_attribute = ESPNOW_ATTRIBUTE_POWER,
                    .responder_value_i   = status,
                };

                espnow_ctrl_data_t data = {
                    .initiator_attribute = ESPNOW_ATTRIBUTE_KEY_1,
                    .responder_attribute = ESPNOW_ATTRIBUTE_POWER,
                    .responder_value_i   = status,
                };

                espnow_ctrl_send(ESPNOW_ADDR_BROADCAST, &data, &frame_head, pdMS_TO_TICKS(1000));
```

poky, 2021-09-26:
Thanks! Will take look for it!

closed 2021-09-26 by poky

poky, 2021-09-28:
Hi, when I check the espnow_frame_head_t struct, the data type of filter_weak_signal is boolean?

https://github.com/espressif/esp-now/blob/9db0e7ff03abda6427d5be517d2a4f44eba340d1/components/espnow/include/espnow.h#L127

zhanzhaocheng, 2021-09-28:
You need to enable filter_weak_signal and set .forward_rssi
[image]

pryhmez, 2023-06-23:
hey zhanahaocheng, please you leave a link to an example on how to use this properly