espressif/esp-now #5: can esp now uses as the repeater on esp3266

ankit-thealchemist, 2021-12-04:
Hello,

I have many esp8266 which does some work  like on off relay, temperature sensor etc. Each esp8266 reports to the wifi network. Problem is each esp might not be in the range of wifi. Can esp now some how help in resolving the issue by extending the range of wifi or helps in relaying the message. Thanks!

ljy770, 2021-12-06:
Yes, esp now can help extending the range of wifi by forwarding the message from device to device. ESP8266 supports esp-now basic APIs like in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_now.html. This  project is developed based on these basic APIs, but haven't been tested on ESP8266. Refer to the project, you can develop your own esp-now project on ESP8266.

ankit-thealchemist, 2021-12-06:
> Yes, esp now can help extending the range of wifi by forwarding the message from device to device. ESP8266 supports esp-now basic APIs like in [...]

Thanks for reply!! I watched the your youtube video and amazed with the capabilities of esp now protocol. I want to replicate the networking using the esp now, from where to start? which mobile appilication you used to control the mesh. Please let me know!!

ljy770, 2021-12-17:
You can try esp-now/examples/control to start. Simple provisioning examples is esp-now/examples/provisioning.
In the video we use [ESP-WIFI-MESH](https://github.com/espressif/esp-mdf) to control the initiator light, app is ESP-Mesh App ([iOS version](https://itunes.apple.com/cn/app/esp-mesh/id1420425921?mt=8) and [Android](https://github.com/EspressifApp/Esp32MeshForAndroid/raw/master/release/mesh.apk)).
The initiator light is provisioned by bluetooth and other lights are provisioned and controlled by the initiator light with esp-now, not by the APP. The examples are not applied now.

ankit-thealchemist, 2021-12-17:
Thanks for info! Closing the issue.

closed 2021-12-17 by ankit-thealchemist