espressif/esp-now #6: Will this be integrated into IDF and ESP8266 NonOS?

gmag11, 2021-12-04:
It would be really nice that this esp-now improvement is included into IDF. Great work!

ljy770, 2021-12-06:
This project is an application developed based on IDF , but will not be integrated into IDF.  You can integrated the esp-now/components to your own IDF project.

sagar448, 2021-12-25:
@ljy770 How can you integrate the ESP-NOW/Components into IDF project? Would it be project level? Or do we drag the components folder into the ESP-NOw folder in the ESP-IDF folder?

closed 2021-12-25 by gmag11

leeebo, 2021-12-25:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

LJYSP, 2022-01-07:
> @ljy770 How can you integrate the ESP-NOW/Components into IDF project? Would it be project level? Or do we drag the components folder into the ESP-NOw folder in the ESP-IDF folder?
like this: 
cp -r ~/esp-now/components/* esp-idf/components/
cp -r ~/esp-now/examples/* esp-idf/examples/