espressif/esp-now #7: Light_Pwm example crashes on esp32

rubikscraft, 2021-12-06:
Hi,

I am trying to get the light pwm example working a normal esp32 but it keeps crashing with the attached log. I am using esp idf 4.4. I also tried turning the wdt off, but that did not help.

Is there anything I'm missing?

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7044
load:0x40078000,len:15548
load:0x40080400,len:4284
0x40080400: _init at ??:?

entry 0x40080680
I (27) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (27) boot: compile time 15:51:56
I (27) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (81) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (89) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (96) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (104) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (111) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (119) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (126) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (134) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (142) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (149) boot: End of partition table
I (153) boot_comm: chip revision: 1, min. application chip revision: 0
I (161) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=1a7f4h (108532) map
I (199) esp_image: segment 1: paddr=0003a81c vaddr=3ffb0000 size=04454h ( 17492) load
I (205) esp_image: segment 2: paddr=0003ec78 vaddr=40080000 size=013a0h (  5024) load
I (207) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8d2b0h (578224) map
I (373) esp_image: segment 4: paddr=000cd2d8 vaddr=400813a0 size=14a6ch ( 84588) load
I (400) esp_image: segment 5: paddr=000e1d4c vaddr=50000000 size=00010h (    16) load
I (411) boot: Loaded app from partition at offset 0x20000
I (412) boot: Disabling RNG early entropy source...
I (423) cpu_start: Pro cpu up.
I (423) cpu_start: Single core mode
I (431) cpu_start: Pro cpu start user code
I (431) cpu_start: cpu freq: 160000000
I (431) cpu_start: Application information:
I (436) cpu_start: Project name:     light_pwm
I (441) cpu_start: App version:      e5a13ba
I (446) cpu_start: Compile time:     Dec  6 2021 15:51:51
I (452) cpu_start: ELF file SHA256:  3dcb7f2f72010677...
I (458) cpu_start: ESP-IDF:          v4.4-dev-3703-gddc44956bf
I (465) heap_init: Initializing. RAM available for dynamic allocation:
I (472) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (478) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (484) heap_init: At 3FFB88E8 len 00027718 (157 KiB): DRAM
I (490) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (497) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (503) heap_init: At 40095E0C len 0000A1F4 (40 KiB): IRAM
I (510) spi_flash: detected chip: generic
I (514) spi_flash: flash io: qio
I (518) esp_core_dump_uart: Init core dump to UART
I (523) cpu_start: Starting scheduler on PRO CPU.
E (543) esp_ota_ops: Rollback is not possible, do not have any suitable apps in slots
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7044
load:0x40078000,len:15548
load:0x40080400,len:4284
0x40080400: _init at ??:?

entry 0x40080680
I (54) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (54) boot: compile time 15:51:56
I (54) boot: chip revision: 1
I (59) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (66) qio_mode: Enabling default flash chip QIO
I (71) boot.esp32: SPI Speed      : 80MHz
I (76) boot.esp32: SPI Mode       : QIO
I (80) boot.esp32: SPI Flash Size : 4MB
W (85) boot.esp32: PRO CPU has been reset by WDT.
W (90) boot.esp32: WDT reset info: PRO CPU PC=0x4008148d
0x4008148d: panicHandler at /home/rubikscraft/esp/esp-idf/components/esp_system/port/panic_handler.c:213

I (96) boot: Enabling RNG early entropy source...
I (102) boot: Partition Table:
I (105) boot: ## Label            Usage          Type ST Offset   Length
I (113) boot:  0 nvs              WiFi data        01 02 0000d000 00008000
I (120) boot:  1 fctry            WiFi data        01 02 00015000 00004000
I (128) boot:  2 log_status       WiFi data        01 02 00019000 00004000
I (136) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (143) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (151) boot:  5 ota_0            OTA app          00 10 00020000 001d0000
I (158) boot:  6 ota_1            OTA app          00 11 001f0000 001d0000
I (166) boot:  7 coredump         Unknown data     01 03 003c0000 00010000
I (173) boot:  8 log_info         Unknown data     01 fe 003d0000 00010000
I (181) boot:  9 reserved         Unknown data     01 ff 003e0000 00020000
I (188) boot: End of partition table
I (193) boot_comm: chip revision: 1, min. application chip revision: 0
I (200) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=1a7f4h (108532) map
I (238) esp_image: segment 1: paddr=0003a81c vaddr=3ffb0000 size=04454h ( 17492) load
I (244) esp_image: segment 2: paddr=0003ec78 vaddr=40080000 size=013a0h (  5024) load
I (246) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8d2b0h (578224) map
I (412) esp_image: segment 4: paddr=000cd2d8 vaddr=400813a0 size=14a6ch ( 84588) load
I (440) esp_image: segment 5: paddr=000e1d4c vaddr=50000000 size=00010h (    16) load
I (451) boot: Loaded app from partition at offset 0x20000
I (451) boot: Disabling RNG early entropy source...
I (462) cpu_start: Pro cpu up.
I (462) cpu_start: Single core mode
I (470) cpu_start: Pro cpu start user code
I (470) cpu_start: cpu freq: 160000000
I (471) cpu_start: Application information:
I (475) cpu_start: Project name:     light_pwm
I (480) cpu_start: App version:      e5a13ba
I (485) cpu_start: Compile time:     Dec  6 2021 15:51:51
I (491) cpu_start: ELF file SHA256:  3dcb7f2f72010677...
I (497) cpu_start: ESP-IDF:          v4.4-dev-3703-gddc44956bf
I (504) heap_init: Initializing. RAM available for dynamic allocation:
I (511) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (517) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (523) heap_init: At 3FFB88E8 len 00027718 (157 KiB): DRAM
I (530) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (536) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (542) heap_init: At 40095E0C len 0000A1F4 (40 KiB): IRAM
I (549) spi_flash: detected chip: generic
I (553) spi_flash: flash io: qio
I (557) esp_core_dump_uart: Init core dump to UART
I (563) cpu_start: Starting scheduler on PRO CPU.
E (582) esp_ota_ops: Rollback is not possible, do not have any suitable apps in slots
```

ljy770, 2021-12-09:
Default light GPIOs are set wrong for ESP32. We will fix it later.
You can modify light GPIOs  as following: or set according to your hardware config.
```
idf.py menuconfig -> Example Configuration-> Light development board selection
    light hardware type (Custom)  --->
(25) Light red pin GPIO number
(26) Light green pin GPIO number
(27) Light blue pin GPIO number
```

rubikscraft, 2021-12-15:
Ah yeah, that was it. I never thought a wrong io config would crash it.

closed 2021-12-15 by rubikscraft