espressif/esp-now #8: espnow_ota_initator_scan hang problem

poky, 2021-12-08:
Hi, I'm trying to test the OTA function, however, I've found that if call espnow_ota_initator_scan function at:

https://github.com/espressif/esp-now/blob/e5a13ba19f12727b9544e1dbd04639bf5016a4d8/components/ota/espnow_ota_initiator.c#L77

It will hang for few seconds, and no other threads can run, is there anyway to get around this problem? Thanks!

ljy770, 2021-12-09:
Blocks happen at xQueueReceive in espnow_recv( wait ticks is not 0). To get around this problem, use  vTaskDelay instead of wait ticks, as following:
```
        vTaskDelay(recv_ticks);
        // for (; espnow_recv(ESPNOW_TYPE_OTA_STATUS, recv_addr, recv_data, &recv_size, &rx_ctrl, recv_ticks) == ESP_OK; recv_ticks = 100) {
        for (; espnow_recv(ESPNOW_TYPE_OTA_STATUS, recv_addr, recv_data, &recv_size, &rx_ctrl, 0) == ESP_OK; (vTaskDelay(100))) {

```
Be careful for other thread process `espnow_recv(ESPNOW_TYPE_OTA_STATUS...)`

poky, 2021-12-09:
> Blocks happen at xQueueReceive in espnow_recv( wait ticks is not 0). To get around this problem, use vTaskDelay instead of wait ticks, as following:
>
> ```
>         vTaskDelay(recv_ticks);
>         // [...]

@ljy770 Thank you for the help and point this out!
This issue may closed

closed 2021-12-09 by poky