espressif/esp-now #9: Broadcasting to our own network only

ankit-thealchemist, 2021-12-11:
Hello,

I was going through the get started example, just want to know how to broadcast message to out own private group. Suppose I have some devices working on esp now and my neighbour would have some devices working on esp now. How the these two network will be separated? Each network will be able to decode only its netwok messages. 
Thanks!

ljy770, 2021-12-14:
You can separate the devices by RSSI and set device group by espnow_send_group().

ankit-thealchemist, 2021-12-14:
@ljy770 Thanks! this sorted out my confusion. Just let me know, when will documentation regarding esp now will be available.
Also just a feature request to also support hash of the password for the authentication.

closed 2021-12-14 by ankit-thealchemist