espressif/esp-now #10: OTA Initator connected to AP with different channel

poky, 2021-12-12:
Hi, I'm trying to play with the OTA example, however, I've found that if the OTA initator is connected to an AP with other channels than default, the responders will not receive any msg.

I've try to add these lines to the frame_head config

https://github.com/espressif/esp-now/blob/e5a13ba19f12727b9544e1dbd04639bf5016a4d8/components/ota/espnow_ota_initiator.c#L86

with
.channel          = ESPNOW_CHANNEL_ALL,

but doesn't help, is there other suggestions I can try out? Thanks!

ljy770, 2021-12-16:
If device has connected to an AP, it will send and receive esp-now message in the AP's channel no matter set ESPNOW_CHANNEL_ALL or not.
ESP-NOW must communicate in the same channel. The OTA responder can be set not connected, and set channel on OTA initiator‘s channel.

poky, 2021-12-16:
@ljy770 Thanks for the reply! I understand that the initiator will send the ESP-NOW packet to the same channel as AP.
I'm trying to solve a real-life case scenario, the responder's channel can not be set as static, because the AP changes the channel randomly once a while, since the responder is not connected to an AP, it's more "free" for responder to change channel, is it possible for the initiator to broadcast it's current channel first, then once responder receives it and change it's channel to the corresponded channel, before the scan list process?

ljy770, 2021-12-16:
Yes, that works.  The responder synchronizes channel with the initiator. For example initiator broadcasts specific message every 1s, and  the responder switches channel every 1s and set channel on which receives the message. If not receive message, the responder will switch channel again.

lhespress, 2023-01-11:
@poky Any update about the issue?

closed 2023-01-11 by poky