espressif/esp-now #11: Hi guys what app are you using ?

ThatBigPrint, 2022-01-06:
hi guys i have started dev on the esp mesh etc but that project looks abandoned? so I'm looking into esp now what app do you guys use it looks like the same as the app for esp mesh ?

LJYSP, 2022-01-07:
In the video we use ESP-WIFI-MESH to control the initiator light, app is also ESP-Mesh App.

ThatBigPrint, 2022-01-07:
Ok thanks for the prompt reply, one other question the iPhone app in GitHub is behind what's released on the app store? can you please send over the updated source code as we need this to transfer changes into our app?

Secondly, has all support for esp mesh stopped as it has not had many updates on git hub?

LJYSP, 2022-01-07:
iPhone app in GitHub : https://github.com/EspressifApp/EspMeshForiOS
By the way, support for esp mesh is not stopped.

ThatBigPrint, 2022-01-07:
yes, I know where the file is but its one version behind the main release on the iPhone store? can you please send me the updated source code for the iPhone?

also what improvements are coming and when will they be released?

fanbaoying, 2022-01-10:
@ThatBigPrint Hi, https://github.com/EspressifApp/EspMeshForiOS The source code here is the latest version, which is synchronized with the App Store.

ThatBigPrint, 2022-01-10:
unfortunately not! just compiled the latest and I still get a prompt on the app to download the latest from the app store ?

ThatBigPrint, 2022-01-11:
any updates on this ?

ThatBigPrint, 2022-01-12:
any updates at all ?

fanbaoying, 2022-01-12:
@ThatBigPrint try again

ThatBigPrint, 2022-01-13:
is the firmware rollback not working on iPhone? and what tools did you use to edit pages looks like vue.js and webkit ?

fanbaoying, 2022-01-13:
WebStorm

ThatBigPrint, 2022-01-13:
what style.css actually, deals with the style of the app? say the footer for instance?

ThatBigPrint, 2022-01-17:
also, the rollback firmware option is missing on iPhone?

ThatBigPrint, 2022-01-17:
do you have any better versions of this app that you have rebranded for other customers? I have spent 2 weeks trying to alter the style and it's just pages and pages of pointless colour calls! if you want customers to take your product on as I want to you need to help them to have at least a presentable app ! or allow mesh integration into rainmaker!

What's nova home?

whats yoti smart 

What's strut lighting?

ThatBigPrint, 2022-01-18:
will i get any help with this or should I start talks with Tuya ?

ThatBigPrint, 2022-01-18:
is there anyone who can email me?

fanbaoying, 2022-01-18:
可以明确一下你的需求吗？我看你还有提到nova home和yoti smart

fanbaoying, 2022-01-18:
但是最开始是基于EspMeshForiOS 项目讨论的

fanbaoying, 2022-01-18:
这些是不同的项目，使用到不同的协议和技术。麻烦明确一下需求，这样更容易帮你解决问题。

ThatBigPrint, 2022-01-18:
the main subject is still esp mesh for ios but it looks like you have all given up on developing for espmesh as both the mdf isn't getting updated too often and the espmesh for ios is not getting updated either? for instance, the android app has firmware rollback and the ios don't and the ios app for iPad is totally broken ?.... i want to know if I'm developing a product that I won't be left behind when you guys officially discontinue espmesh also noticed that you are stopping support for the interactive drag block section for the iPad app further leading towards me thinning you have given up?

主要主题仍然是适用于 ios 的 esp 网格，但看起来你们都放弃了为 espmesh 开发，因为 mdf 没有经常更新，而适用于 ios 的 espmesh 也没有更新？例如，android 应用程序有固件回滚，而 ios 没有，而 iPad 的 ios 应用程序完全坏了？....我想知道我是否正在开发一种产品，当你在伙计们正式停止 espmesh 还注意到你停止支持 iPad 应用程序的交互式拖动块部分进一步导致我瘦你已经放弃了？

fanbaoying, 2022-01-18:
@ThatBigPrint  We have test the firmware rollback function on ESP32 with iphone, it works，please see the attached log.
[image]
If the firmware rollback is not working on your device, could you please submit more details about the issue? Chip/module model, MDF commit version, iphone model, and related log will be useful. 
The MDF team will continue provide support, if you meet issue, please submit through: https://github.com/espressif/esp-mdf/issues.

ThatBigPrint, 2022-01-18:
Hi, the option is not there in the iPhone app? it's there on the android but not iPhone when I hold down on the light and the menu comes up saying options like update firmware and reset device etc.... the option to rollback firmware is missing... but it is there on android?

and this option is missing on iPad tested on app store release on iPad pro  please see the link below referencing time stamp in your youtube demo https://youtu.be/PZAvPKGhZ8Y?t=105

I already have and no one replies.... are you implementing mesh on rainmaker as I want to use that but my issue is I will have over 60 devices on a network hence I'm turning to mesh if you have a solution that I can use rainmaker that has mesh function please let me know now rather than me waste more weeks trying to fix countless errors ith esp mesh app....

I'm trying my best to stay with esp... but its becoming more difficult to justify

ThatBigPrint, 2022-01-19:
any updates on this?

lhespress, 2023-01-11:
@fanbaoying Could you give some feedback about this?

ThatBigPrint, 2023-03-13:
back again the ios release is still not on github ! the versionRollback option is not in the index.html but the app on the store has the button ???

closed 2024-01-31 by lhespress