espressif/esp-now #12: provisioning example problem

poky, 2022-01-08:
Hi, I have problem trying the provisioning example, the initiator works fine and able to retrieve the ssid/pass from the responder for the first 3 tries (manually resets for the testing purpose) , but after about 4 or 5 times, it shows the following warning and unable to retrieve ssid info, does anyone know what the problem might be? Thanks!

[image]

LJYSP, 2022-01-14:
It seems that countdown is error in provisioning_responder function, you can try this：
```
    for (int32_t start_ticks = xTaskGetTickCount(), recv_ticks = pdMS_TO_TICKS(30 * 1000); recv_ticks > 0;
            recv_ticks = pdMS_TO_TICKS(30 * 1000) - (xTaskGetTickCount() - start_ticks)) {
```

lhespress, 2023-01-11:
@poky I see you poky reacted with thumbs up emoji, so it has resolve your issue, right? Please close the issue if yes, thanks.

closed 2023-01-11 by poky