espressif/esp-now #13: Is this compatible with ESP-IDF release/v4.1 and release/v3.3?

sagar448, 2022-01-13:
We would like to integrate this into our projects which are currently based on release 4.1 and 3.3. Would this be compatible with those versions?

Thanks!

LJYSP, 2022-01-14:
This project is compatible with release v4.3, not compatible with release 4.1 and 3.3. Because some apis are not the same, like console  api in components/debug, led api in examples/dev_kits/common_components/led.

what functions do you will use in espnow, you may try to change the api to your  IDF version.

Raghav3107, 2022-03-22:
Hello @LJYSP 

Does this project work with the idf v4.2.2 ? Can we directly use the components ? 
Does the Components utils are based on the v4.3 ??

lhespress, 2023-01-11:
@sagar448 Please refer to the instructions of [idf-version](https://github.com/espressif/esp-now/#idf-version) and checkout it to the master branch.

closed 2023-04-14 by lhespress