espressif/esp-now #15: Developing a many-to-many network?

nickkoza, 2022-01-24:
Hello! I've watched the official video on ESP-Now (https://www.youtube.com/watch?v=QmvMtgNs9r8) and have a few questions about the details:

* The example in the video shows a one-to-many relationship between devices, where a single keypad is changing 255 lights. **Is it possible to design the network so that the nodes are all peer to each other and any node can send a message to any other node?**

* **How many devices can be paired together into a many-to-many network?** The video shows 255 lights but the documentation talks about keeping peers under 20 devices.

* **Can new devices be paired with the entire network at once, or do I have to pair the new device individually with each existing device?** As in, can a new device be paired once and quickly become "aware" of the entire network of devices and start sending messages to them even if they weren't directly paired?

* At 4 minutes 22 seconds in the video they mention developing a multi-hop network, but I haven't been able to find anything about this in the documentation. **How does the multi-hop network functionality work?**

* Lastly, at 6 minutes and 31 seconds they show a web interface for debugging. **Is it possible to get that web interface?** I don't see the source for it anywhere.

Thanks for your help!

LJYSP, 2022-01-25:
1. The keypad sends broadcast packets to lights, and the lights can also forward the message to more lights.

2. If send unicast packets to a device, the sender should add the device to peer and keeping peers under 20 devices. If not send unicast packet, like in the project, mainly in broadcast packet, the sender just need add broadcast peer and send broadcast to many lights.

3. Devices can be paired, we called bind in the project, for example example/control. Or not paired, a new device can directly send to or recv from other devices.

4.`Readlize a muti-hop network through data forwarding function`. Please refer to esp-now/examples/dev_kits/button_remote_control

5. Please refer to esp-now/examples/wireless_debug, open "Enable WebServer debugging" in menuconfig. Connect to the softap and access to the web server `espnow-webserver/`.

nickkoza, 2022-01-26:
Okay, excellent. That makes sense. I was wondering if those examples were using broadcast but the video hadn't mentioned. Thanks for your help!

closed 2022-01-26 by nickkoza

JuandreG, 2022-05-08:
Can not find the "esp-now/examples/dev_kits/button_remote_control" mentioned. Can we please have a new link?

larryli, 2022-08-23:
> Can not find the "esp-now/examples/dev_kits/button_remote_control" mentioned. Can we please have a new link?

https://github.com/espressif/esp-now/tree/1e7a050d92a88eb45ab03c2c92af9838454791d4/examples/dev_kits/button_remote_control