espressif/esp-now #16: ESP32 Sensor network with ESPNOW protocol

oliverrr97, 2022-01-26:
I would like to ask about the project on this link

https://github.com/espressif/esp-now/tree/master/examples

I would like to create a project that combines the examples on this link. In other words, I want to create a sensor network from multiple ESP32 using the ESPNOW protocol. Could you please help me with a detailed description of examples on this line or possibly with their connection to a larger project where I will be able to communicate more ESP32 and send some data at regular intervals.
Please do not know how to add at least detailed comments to the examples on that link?

oliverrr97, 2022-01-26:
I would like to ask if anyone could help me with the detailed commenting of these codes and the subsequent linking of these examples into one more complex one.

I need to achieve a larger program where a sensor network in the ESPNOW protocol will be created using ESP32 ... 4-5 ESP32 devices should communicate with each other in this network. one main ESP32 will be plugged into the PC and the other will be on battery or mains. They have to communicate with each other to send data and after receiving and subsequent sending, the ESP32 should go into sleep mode and wake up the next time it is received and sent.

LJYSP, 2022-01-26:
You may get started with the basic sending and receiving example  in esp-now/examples/get-started. README is under each example.
We haven't given the sleep mode in the project, but  we will help to apply the method.  If the other devices only communicate with the main?  If so, the other devices can go to deep sleep mode when no data to transmit.  Deep sleep please refer to [example](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/system/deep_sleep) and [docs](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-reference/system/sleep_modes.html?highlight=deep_sleep)

lhespress, 2023-01-11:
@oliverrr97 Any update about this?

closed 2023-04-14 by lhespress