espressif/esp-now #17: Problem in connecting ESPNOW to BLYNK

amir-eshaqy, 2022-02-10:
Hi, I want to send an int value from ESP 8266 to ESP 32 using the ESP NOW protocol.
But when I use this protocol (**ESP NOW**) in (((**blynk**))) I can not find any value

This is my ESP 32(Is connected to blynk) code
https://pastebin.com/pzLcsNtn

This is my ESP 8266code
https://pastebin.com/3MxVxrb5

I can not access the variable sent by ESP 8266 when I use BLYNK
I’m sure my ESP NOW protocol is working properly because when I use the following code in ESP 32 everything is fine

Note: I can send commands from ESP 32 to ESP 8266 with this protocol. But I can not do the contrary
Note: If you want to test the code, change the MAC address and BLYNK_TEMPLATE_ID

ESP code 32 without using Blink which works properly (ESPNOW protocol only)

https://pastebin.com/z3DQU7LD

Please help me

LJYSP, 2022-02-21:
Try this: set 20M bandwidth in ESP32 , and see if it works.
    `ESP_ERROR_CHECK(esp_wifi_set_bandwidth(WIFI_IF_STA, WIFI_BW_HT20));`

lhespress, 2023-01-11:
@amir-eshaqy Any update about this?

closed 2023-04-14 by lhespress