espressif/esp-now #18: Esp 8266

renanmbs, 2022-03-06:
I am trying to have a duplex communication with an esp32 and an esp8266-01.

Is it supported? Or just communications between esp32s?

LJYSP, 2022-03-07:
ESP8266 supports esp-now basic APIs like in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_now.html. This project is developed based on these basic APIs, but haven't been ported on ESP8266. Refer to the project, you may develop your own esp-now project on ESP8266.

lhespress, 2023-01-11:
@renanmbs Any update about this?

renanmbs, 2023-01-11:
I was able to send some bytes to xFFFF, and the esp32 received it. But I wasn’t able to do a duplex communication as I was intending

lhespress, 2023-01-12:
@renanmbs Thanks for your feedback, i'll check it again.

lhespress, 2023-04-14:
@renanmbs The master branch of ESP8266 has fix the issue, please update and try again.

closed 2023-06-02 by lhespress