espressif/esp-now #19: espnow_deinit example request

poky, 2022-03-08:
Hi, I'd like to call the espnow_deinit() to release the resources once I'm done with the protocol
Do you have a working example to show how it can be done?
If I call it directly, it will panic

`assert failed: vQueueDelete queue.c:2140 (pxQueue)`

Caused by this line

https://github.com/espressif/esp-now/blob/02d41754038cc727d3dfabedd1e3df9e2c21d6ea/components/espnow/src/espnow.c#L795

LJYSP, 2022-03-09:
Thanks for the question, we will try to fix it and tell you when test it ok.

LJYSP, 2022-03-09:
@poky you can try the following espnow_deinit:
```
esp_err_t espnow_deinit(void)
{
    ESP_ERROR_RETURN(!g_espnow_config, ESP_ERR_ESPNOW_NOT_INIT, "ESPNOW is not initialized");

    /**< De-initialize ESPNOW function */
    ESP_ERROR_CHECK(esp_now_unregister_recv_cb());
    ESP_ERROR_CHECK(esp_now_unregister_send_cb());
    ESP_ERROR_CHECK(esp_now_deinit());

    for (int i = 0; i < ESPNOW_TYPE_MAX; ++i) {
        if (g_espnow_queue[i]) {
            uint8_t *tmp_data = NULL;

            while (xQueueReceive(g_espnow_queue[i], &tmp_data, 0)) {
                ESP_FREE(tmp_data);
            }

            vQueueDelete(g_espnow_queue[i]);
            g_espnow_queue[i] = NULL;
        }
    }

    ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, wifi_event_handler));

    vSemaphoreDelete(g_send_lock);
    g_send_lock = NULL;

    vEventGroupDelete(g_event_group);
    g_event_group = NULL;

    ESP_FREE(g_espnow_config);
    g_espnow_config = NULL;

    return ESP_OK;
}
```

poky, 2022-03-09:
Hi, thanks for the reply!
I've tried the code, but now it shows the following exception:

`assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)`

From this line:

https://github.com/espressif/esp-now/blob/60961c6127b3dae7776b93865e9be6ca2c26121d/components/espnow/src/espnow.c#L626

LJYSP, 2022-03-09:
Does it show exception everytime? Can you show the log before exception?
After espnow_deinit, the forward task will be deleted.

poky, 2022-03-09:
Hi, I'm using the provisioning example to test the deinit function, set as an initiator, and add a delay for the task to finish then call deinit, here's what the code looks like:

[image]

After 30s of the task, it's panic and show following messages:

assert failed: vQueueDelete queue.c:2140 (pxQueue)

Backtrace:0x40081a62:0x3ffbaa500x400884c5:0x3ffbaa70 0x4008f605:0x3ffbaa90 0x4008968e:0x3ffbabb0 0x400d726b:0x3ffbabd0 0x400d60e3:0x3ffbac00 0x4013ff39:0x3ffbac50 0x4008bbfe:0x3ffbac70 
0x40081a62: panic_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/panic.c:402

0x400884c5: esp_system_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/esp_system.c:121

0x4008f605: __assert_func at /Users/lee/espidf/4.4/esp-idf/components/newlib/assert.c:85

0x4008968e: vQueueDelete at /Users/lee/espidf/4.4/esp-idf/components/freertos/queue.c:2140 (discriminator 1)

0x400d726b: espnow_deinit at /Users/lee/Desktop/esp-now/components/espnow/src/espnow.c:797 (discriminator 2)

0x400d60e3: app_main at /Users/lee/Desktop/esp-now/examples/provisioning/build/../main/app_main.c:142

0x4013ff39: main_task at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x4008bbfe: vPortTaskWrapper at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/xtensa/port.c:131

The line 797 is refer to the line of:

`vQueueDelete(tmp_queue);`

LJYSP, 2022-03-11:
> Hi, thanks for the reply! I've tried the code, but now it shows the following exception:
>
> `assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)`
>
> From this [...]

For this error, can you show the example code and the log before exception?

LJYSP, 2022-03-11:
> Hi, thanks for the reply! I've tried the code, but now it shows the following exception:
>
> `assert failed: spinlock_acquire spinlock.h:122 (result == core_id || result == SPINLOCK_FREE)`
>
> From this [...]

Please modify [this line](https://github.com/espressif/esp-now/blob/60961c6127b3dae7776b93865e9be6ca2c26121d/components/espnow/src/espnow.c#L625) as follow and check:
```
    while (g_espnow_config && g_espnow_queue[ESPNOW_TYPE_FORWARD]) {
```

poky, 2022-03-14:
Hi, after change, it still crash with following outputs

I (639) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (739) wifi:mode : sta (d8:a0:1d:5d:52:f4)
I (749) wifi:enable tsf
I (749) ESPNOW: espnow [version: 1.0] init
I (749) espnow_prov: Responder beacon start, timer: 30s
I (30849) espnow_prov: Responder beacon end

assert failed: vQueueDelete queue.c:2140 (pxQueue)

Backtrace:0x40081a62:0x3ffbaa500x400884c5:0x3ffbaa70 0x4008f605:0x3ffbaa90 0x4008968e:0x3ffbabb0 0x400d723b:0x3ffbabd0 0x400d60e3:0x3ffbac00 0x4013ff09:0x3ffbac50 0x4008bbfe:0x3ffbac70 
0x40081a62: panic_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/panic.c:402

0x400884c5: esp_system_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/esp_system.c:121

0x4008f605: __assert_func at /Users/lee/espidf/4.4/esp-idf/components/newlib/assert.c:85

0x4008968e: vQueueDelete at /Users/lee/espidf/4.4/esp-idf/components/freertos/queue.c:2140 (discriminator 1)

0x400d723b: espnow_deinit at /Users/lee/Desktop/esp-now/components/espnow/src/espnow.c:798 (discriminator 2)

0x400d60e3: app_main at /Users/lee/Desktop/esp-now/examples/provisioning/build/../main/app_main.c:142

0x4013ff09: main_task at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x4008bbfe: vPortTaskWrapper at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/xtensa/port.c:131

LJYSP, 2022-03-14:
> 

Please update ESP_NOW to the lastest version and have a try.

poky, 2022-03-14:
Hi, have the latest version updated, the deinit works fine for the first time, but when init esp now again, the following messages:

I (31747) espnow_prov: Responder beacon start, timer: 30s

assert failed: spinlock_acquire spinlock.h:123 ((result == SPINLOCK_FREE) == (lock->count == 0))

Backtrace:0x40081a62:0x3ffc77500x400884c5:0x3ffc7770 0x4008f605:0x3ffc7790 0x4008bde1:0x3ffc78b0 0x4008925b:0x3ffc78d0 0x400d6618:0x3ffc7910 0x4008bbfe:0x3ffc7940 
0x40081a62: panic_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/panic.c:402

0x400884c5: esp_system_abort at /Users/lee/espidf/4.4/esp-idf/components/esp_system/esp_system.c:121

0x4008f605: __assert_func at /Users/lee/espidf/4.4/esp-idf/components/newlib/assert.c:85

0x4008bde1: spinlock_acquire at /Users/lee/espidf/4.4/esp-idf/components/esp_hw_support/include/soc/spinlock.h:123
 (inlined by) xPortEnterCriticalTimeout at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/xtensa/port.c:288

0x4008925b: vPortEnterCritical at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) xQueueReceive at /Users/lee/espidf/4.4/esp-idf/components/freertos/queue.c:1400

0x400d6618: espnow_forward_task at /Users/lee/Desktop/esp-now-master/components/espnow/src/espnow.c:626

0x4008bbfe: vPortTaskWrapper at /Users/lee/espidf/4.4/esp-idf/components/freertos/port/xtensa/port.c:131

lhespress, 2023-01-11:
@poky Please update ESP-NOW to the master branch and try again.

closed 2023-01-11 by poky