espressif/esp-now #20: Problem with flash hello_world.bin after reset in OTA

oliverrr97, 2022-03-15:
I have Problem with flash hello_world.bin after reset in OTA. Responder dont flash hello_world after received
[image]
.
[image]
[image]

LJYSP, 2022-03-16:
@oliverrr97  
Please download the hello_word.bin directly in ESP32 and check if it runs OK ?

oliverrr97, 2022-03-16:
yes, it runs OK.

LJYSP, 2022-03-16:
@oliverrr97  
I have reproduced the same problem on v4.3.2. We will try to fix it and tell you when it is OK. You may try with v4.3.1 and ota example is tested OK on it .

LJYSP, 2022-03-16:
@oliverrr97  Please set  “CONFIG_ESPTOOLPY_FLASHFREQ_80M=y” in sdkconfig.default when build  hello_word.bin on ESP32.(For ESP32, the default flash frequency is 40M, For other chip, it is 80M). 
In ota example，the default flash frequency is 80M in sdkconfig.defaults.  The frequency should keep the same.

closed 2023-01-11 by lhespress