espressif/esp-now #21: Problem with debug with write http data

oliverrr97, 2022-03-22:
i have problem in example "debug". here is screenshots of my problem.
[image]
[image]

LJYSP, 2022-03-24:
@oliverrr97  Please use a TCP testing tool (any third-party TCP testing software) to create a TCP server.

oliverrr97, 2022-03-30:
I used a testing tool to create an http server, but I still don't know how to enter commands there: help, system_info ...
[image]

LJYSP, 2022-03-31:
Please press "enter " in the left console and the output liike this " esp32c3>"
Then input the command "help" and press "enter", the output liike this:
```
esp32c3> help
help 
  Print the list of registered commands

version 
  Get version of chip and SDK
...
```

oliverrr97, 2022-03-31:
I am not allowed to write anything on the left console. even when I press enter nothing happens. I use the ESP32 DevKitc V4 model.

LJYSP, 2022-04-01:
What IDF version do you use?

oliverrr97, 2022-04-01:
I tried it on version 4.3.2 and version 4.4

oliverrr97, 2022-04-06:
what can i do with it?

LJYSP, 2022-04-11:
I don't know why, but you can try IDF/example/wifi/iperf example to test if input command is ok. Or change a other ESP32 board to test.

lhespress, 2023-01-11:
@oliverrr97 Any update about this?

closed 2023-04-14 by lhespress