espressif/esp-now #22: Problem with example control with led_strip.h

oliverrr97, 2022-03-22:
I can't find the header file led_strip.h. this header file is neither in the components nor in the esp-idf folder

LJYSP, 2022-03-22:
Please build with ESP-IDF tag v4.4. That version supports downloading button and led component automatically from  https://staging.components.espressif.tools/search/org:espressif

oliverrr97, 2022-03-22:
How should I do that

oliverrr97, 2022-03-22:
I use the development board esp32 devkitc V4, is it possible to demonstrate an example of control on this board?
[image]
no bind is created when the boot button is pressed two times

LJYSP, 2022-03-23:
> I use the development board esp32 devkitc V4, is it possible to demonstrate an example of control on this board? [image] no bind is created when the boot button is pressed two times

Does this example build with  esp-now v1.0 ？ In that version example, you may press the boot button two times in a short time to bind.
But in your log, it seems it is version before esp-now v1.0 or you add the log?. Anyway, in old version, bind is to press the boot button for more than 1 second.

oliverrr97, 2022-03-23:
its okey this problem is solved. i have another problem yet with debug.

closed 2023-01-11 by lhespress