espressif/esp-now #23: Problem with wireless_debug

oliverrr97, 2022-03-24:
please how do i connect two esp32 devices in this example? how to insert the added pairing code there? is it necessary to insert the added code into both devices? how do you set menucofig? from that tutorial it's not clear to me how to pair two devices.

LJYSP, 2022-03-25:
You may  not need to connect the two esp32 devices. One device runs the debug example, the other runs this example, and this  example can monitor the debug example device. Please refer to  examples/wireless_debug/README.md.
What is the added pairing code?

oliverrr97, 2022-04-06:
[image]
there is a wireless debug device on the left side and a debug device on the right ... the list at the bottom of the left page is displayed after pressing the button to refresh the list on the mobile ... but no device will appear in the list on the mobile

lhespress, 2023-01-11:
@oliverrr97 Do you try it on master branch and have any update about this?

closed 2023-04-14 by lhespress