espressif/esp-now #25: Is API in Programming Guide old version of ESP-Now?

NostalgistL00, 2022-03-28:
I wanna use esp-now in my little project of ESP8266,i try to find reference in programming guide,and later i saw this.
the API is not the same ,so is this new version of ESP-Now?If so,is there any examples of old version.
Thanks

LJYSP, 2022-03-29:
This esp-now project is application of esp now, based on the api defined in IDF esp_now.h.  Please refer to IDF/example/wifi/espnow to use the basic esp-now apis.

NostalgistL00, 2022-03-29:
> 

Thank you.
This helps a lot. :)

NostalgistL00, 2022-03-29:
> Thank you.
> This helps a lot. :)

> This esp-now project is application of esp now, based on the api defined in IDF esp_now.h. Please refer to IDF/example/wifi/espnow to use the basic esp-now apis.

Thank you.
This helps a lot. :)

NostalgistL00, 2022-03-29:
> This esp-now project is application of esp now, based on the api defined in IDF esp_now.h. Please refer to IDF/example/wifi/espnow to use the basic esp-now apis.

Thank you.
This helps a lot. :)

closed 2023-01-11 by lhespress