espressif/esp-now #26: Ota

oliverrr97, 2022-03-29:
I would like to ask about the ota project. This project works in such a way that if we upload the hello world to the responder, for example, it is no longer possible to update the responder to another project?

can a new uploaded program run on the responder at once so that each time the responder responds, the bootloader looks to see if an update from the initiator is available?

LJYSP, 2022-03-30:
hello_world.bin is just used for test. You may upload a new firmware which is include ota responder function, for example a little different ota responder example. when reboot to new image, it can also respond the initiator.

oliverrr97, 2022-03-30:
can you advise me how to combine get-started with example ota responder?

LJYSP, 2022-03-31:
We are now restructuring the project to combine different functions. Now combining different functions may need a  little change.
To combine get-started with example ota responder, you may add the following code to get-started example, and  add examples/ota/sdkconfig.defaults  partitions_4MB.csv to examples/get-started/
```
    espnow_ota_config_t ota_config = {
        .skip_version_check       = true,
        .progress_report_interval = 10,
    };
    espnow_ota_responder_start(&ota_config);
```
Change the wait ticks to avoid block in one of the espnow_recv  :
```
//components/ota/espnow_ota_responder.c Line:404
        ret = espnow_recv(ESPNOW_TYPE_OTA_DATA, src_addr, data, &size, NULL, 100);
        ESP_ERROR_CONTINUE(ret != ESP_OK, "");

```
```
//examples/get-started/main/app_main.c Line: 102
        ret = espnow_recv(ESPNOW_TYPE_DATA, addr, data, &size, &rx_ctrl, 100);
        ESP_ERROR_CONTINUE(ret != ESP_OK, "");

```

oliverrr97, 2022-04-07:
can you please describe to me in more detail what needs to be done. because it is certainly not enough to make only these conclusions that you have written to me. please i would really need it to work with the update at any time. at least on a simple example.

oliverrr97, 2022-04-11:
I've already made it. why does the responder have to be connected to the router? shouldn't he communicate with the initiator via ESPNOW?

LJYSP, 2022-04-12:
It's just a simple way to make sure the responder  channel the same with the initiator channel.

LJYSP, 2022-04-12:
The responder and the initiator connect to the same AP and they will on the same channel which the AP is working on.

oliverrr97, 2022-04-12:
can you give me instructions on how to set it up

closed 2023-04-14 by lhespress