espressif/esp-now #28: Bind Limit of 20

tiki30, 2022-04-03:
Hi, can you explain exactly what limits the pairing of devices to a maximum of 20 ?. If I need to communicate with more than 20 devices without using broadcast, is there a mechanism to do this ?
Many Thanks.

LJYSP, 2022-04-06:
The WIFI lib and memory cost limit the pairing of devices to a maximum of 20.
If I need to communicate with more than 20 devices without using broadcast, is there a mechanism to do this ?
==> You can add peer address before sending and delete peer address after sending.
```
espnow_add_peer(src_addr, NULL);
espnow_send();
espnow_del_peer(src_addr);
```

svdrummer, 2022-12-31:
If I broadcast would there be a limit. I can set wifi to 15, however, i am switching to esp_now.
I am thinking broadcast is just to one peer FF's in mac.
I dont care about the result of the send

lhespress, 2023-01-06:
You can ignore the limit of 20 in broadcast mode. We have tested 100 devices as responder which have the max delay timer is 315ms.

closed 2023-04-14 by lhespress