espressif/esp-now #33: ESP-Now Mesh

JuandreG, 2022-05-08:
Hello.

I found out about ESP-Now a few weeks ago and since then I've been testing out the functionality with great success. This mode of communication with the ranges that can be achieved opens a lot of doors for new projects.

Now to further extend the range one would probably need a type of repeater node between nodes that send data and nodes that receive data. And if you then want 2 way communication between the start and end nodes the picture starts looking like a mesh network.

Is mesh networking possible with ESP-Now? And if so how would one build this network of ESP-Now devices?

[image]

I was able to establish 2 way communication with Sender devices and Receiver device without defining mac addresses on any devices and also without the Repeater devices. I figured that defining mac addresses on any device would be nightmare to maintain if any of the devices were to break. By using the SSID's of the devices to establish a connection I think is a better way to go.

So my question is:

Can Receiver then send a message back to Sender 1 or Sender 2 using a Repeater without defining the mac address path to use?

JuandreG, 2022-07-08:
Any feedback?

LJYSP, 2022-07-11:
Sorry for being late in replying. ESP-NOW is a kind of connectionless Wi-Fi communication protocol. I think `ESP-WIFI-MESH` may meet your needs. Please refer to [ESP-WIFI-MESH](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/esp-wifi-mesh.html), and the [project](https://github.com/espressif/esp-mdf).