espressif/esp-now #35: in ota   all device connect to same ap or just initiator connect internet and send ota via esp-now

mrdsn3, 2022-05-15:
I was a little confused
In Exempla OTA, in order to connect to the initiator, both devices must be connected to an access point, or it is enough for the initiator to download the file and then send it to the devices via Broadcast or Unicast.

leeebo, 2022-05-15:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

LJYSP, 2022-07-06:
@mrdsn3  The responder devices may not connect to an access point,  but should be at the same channel with initiator. Because initiator connects to an AP and can only send esp-now in the AP channel.
You can set the channel of the access point to channel 1, or set responder device channel to the AP  channel(by esp_wifi_set_channel()).

closed 2023-04-14 by lhespress