espressif/esp-now #37: ESP long range mode persists on flash, erase, and reset!!!

zachtos, 2022-05-26:
I spent 2 weeks struggling to figure out that ESP32 if you turn on long range mode, it will break your wifi, even after your erase the chip and put new firmware on that is not in long range mode.  It will persist on power cycle, I feel this is NOT OBVIOUS!!!!  it needs to be addressed in the manual or fixed.

```
WiFi.mode(WIFI_STA);
esp_wifi_set_protocol( WIFI_IF_STA, WIFI_PROTOCOL_LR ); //this will double range but half packet rate
```

and to fix the problem, even after turn off/on, before you attempt to use wifi again, you MUST do this

```
WiFi.disconnect(true); //required to change modes
WiFi.mode(WIFI_STA);
esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);
```

LJYSP, 2022-05-27:
@zachtos @LaserTagMods  Please test with[ IDF espnow example](https://github.com/espressif/esp-idf/tree/v4.4.1/examples/wifi/espnow) and enable ESPNOW_ENABLE_LONG_RANGE, idf.py menuconfig-> Example Configuration ->Enable Long Range ->y.

lhespress, 2023-01-11:
@zachtos Any update about this.

lhespress, 2023-04-14:
@zachtos Closing this issue since there has been no update on this. Please feel free to reopen if required.

closed 2023-04-14 by lhespress

fritzlb, 2023-08-17:
Thank you so much for this info! I couldn’t figure out why suddenly my WiFi stopped working after testing esp-now, but this solves it.