espressif/esp-now #38: espNow crashing with dmd32 lib

microElabDevelop, 2022-06-20:
hi, im using dmd32 lib with esp now ,, but simple program rebooting again and again,
[image]
`#include <esp_now.h>
#include <WiFi.h>

#include <DMD32.h> //
#include "fonts/SystemFont5x7.h"
#include "fonts/Arial_black_16.h"

//Fire up the DMD library as dmd
#define DISPLAYS_ACROSS 1
#define DISPLAYS_DOWN 1
DMD dmd(DISPLAYS_ACROSS, DISPLAYS_DOWN);
//Timer setup//create a hardware timer of ESP32
hw_timer_t * timer = NULL;

//////////////
typedef struct struct_message{
uint8_t a;
uint8_t b;

}struct_message;
// Create a structured object
struct_message myData;

void OnDataRecv(const uint8_t * mac, const uint8_t incomingData, int len);
/--------------------------------------------------------------------------------------
Interrupt handler for Timer1 (TimerOne) driven DMD refresh scanning, this gets
called at the period set in Timer1.initialize();
--------------------------------------------------------------------------------------*/

void IRAM_ATTR triggerScan()
{
dmd.scanDisplayBySPI();
}
void setup(void){
Serial.begin(115200);
Serial.println("receive");

// return the clock speed of the CPU
uint8_t cpuClock = ESP.getCpuFreqMHz();
// Use 1st timer of 4
// devide cpu clock speed on its speed value by MHz to get 1us for each signal of the timer
timer = timerBegin(0, cpuClock, true);
// Attach triggerScan function to our timer
timerAttachInterrupt(timer, &triggerScan, true);
// Set alarm to call triggerScan function
// Repeat the alarm (third parameter)
timerAlarmWrite(timer, 300, true);

// Start an alarm
timerAlarmEnable(timer);

//clear/init the DMD pixels held in RAM
dmd.clearScreen( true ); //true is normal (all pixels off), false is negative (all pixels on)
///////////////
// Set ESP32 as a Wi-Fi Station
WiFi.mode(WIFI_STA);
// Initilize ESP-NOW
if (esp_now_init() != ESP_OK) {
Serial.println("Error initializing ESP-NOW");
return;
}
// Print MAC Address to Serial monitor

Serial.print("Responder ");
Serial.print("MAC Address: ");
Serial.println(WiFi.macAddress());

// Register callback function
esp_now_register_recv_cb(OnDataRecv);
}

/--------------------------------------------------------------------------------------
loop
Arduino architecture main loop
--------------------------------------------------------------------------------------/
unsigned int count;
char buf[]="1234";
void loop(void){

byte b;
// 10 x 14 font clock, including demo of OR and NOR modes for pixels so that the flashing colon can be overlayed
dmd.clearScreen( true );
dmd.selectFont(Arial_Black_16);
snprintf(buf, sizeof(buf), "%d",count);
dmd.drawString( 3,1,buf,strlen(buf), GRAPHICS_NORMAL );

delay(500);
count++;
}

// Callback function executed when data is received
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {
memcpy(&myData, incomingData, sizeof(myData));
Serial.print("Data received: ");
Serial.println(len);
Serial.print("Character Value: ");
Serial.println(myData.a);
Serial.print("Integer Value: ");
Serial.println(myData.b);
}`

LJYSP, 2022-08-29:
In the log, `PSRAM  ID read error,` please check.

closed 2023-04-14 by lhespress

Axliyor, 2024-09-29:
dmd32 with wifi is not working at the same time