espressif/esp-now #41: use ESP-now and ADC at the same time

RTLhamster, 2022-08-16:
Hi there,

I try to send unidirectional ESP32 ADC2  (and other) data continuously to a second ESP32.
The ADC does not return data if ESP-now is enabled.
As far as I know is that ADC2 is used by the Wi-Fi driver and must be disabled to read ADC samples.

I tried to disable wifi:
` esp_wifi_stop();`
read data:
` esp_err_t r = adc2_get_raw(ADC2_CHANNEL_3, ADC_WIDTH_12Bit, &data[0]);`    
and re-start the WiFi front-end
` esp_wifi_start();`

It works but it takes quite some time.
Without stopping/starting WiFi I can easily meet my timing constrain of sending data every 10ms. With it, it needs at least 70ms which is too much for my application.

Is it possible to disable/enable the front-end differently/faster?

Thanks!

LJYSP, 2022-08-29:
You may try to close the log during  wifi stop and wifi start，like using ` esp_log_level_set("wifi", ESP_LOG_NONE);` ` esp_log_level_set("phy_init", ESP_LOG_NONE);`

RTLhamster, 2022-09-05:
Hi LJYSP,
thanks for your response.

I've tried your suggestion, but it didn't decrease the loop-time.
For me its not clear if and why the disabling and re-enabling of the wifi/esp_now stack is required.
There was a way to store/re-store the ADC config.:
https://github.com/espressif/arduino-esp32/issues/440#issuecomment-1008165291
but it doesn't work.

Maybe some could ague that the WiFi stack could be messed up while doing it.
But as far as I understand esp-now, its a connection less protocol:
sending data, wait for reply, re-configure ADC should not result in any issue here.
Also with respect of low-power, esp-now API should give us a easier and fast way to enable/disable it.

LJYSP, 2022-09-07:
ESP-NOW uses wifi action frame to send  and receive data, so wifi start is needed. 

Or you may try ESP32-S2, it can use wifi while using ADC2.

lhespress, 2023-01-11:
@RTLhamster Any update about this?

RTLhamster, 2023-01-11:
> 
Hi @lhespress,
unfortunately not, we change the PCB to use ADC1 GPIOs.
It's a pity since a software support on ESP-now could be added I believe... 
But we have to live with it :/

lhespress, 2023-01-12:
Thanks for your feedback, i'll check it again.

lhespress, 2023-04-24:
@RTLhamster I reproduce the issue on ESP32, we didn't suggest that use ADC2 when you need `WIFI` or `BLE` because of the `PHY` also used it.

closed 2023-06-02 by lhespress