espressif/esp-now #42: How to setup in PlatformIO

brskt-dev, 2022-08-22:
Hello

I use PlatformIO for my ESP32 
projects, with arduino-espressif32.
I'm trying to find a way to filter Esp Now connections somehow related to signal strength, and then I find this repo...

What I need to understand is how to setup this esp-now repo in my projects, to use the "filter_weak_signal" option!

I hope someone help me :/

leeebo, 2022-08-22:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

LJYSP, 2022-08-29:
You can copy `esp-now/components/` to  `esp-idf/components/,`  to use esp-now as components of esp-idf.

brskt-dev, 2022-09-01:
> You can copy `esp-now/components/` to  `esp-idf/components/,`  to use esp-now as components of esp-idf.

I'll try this, thank u 🙏🏼

closed 2022-09-01 by brskt-dev