espressif/esp-now #45: Having esp-now run on Linux (like Raspberry Pi).

nielsnl68, 2022-10-04:
I really like the esp-now protocol. At this moment you can only talk between esp's and for some long time i would love to see a solution to my raspberry pi as well.

Therefor i have a small request could you build something that allows the PI directly to the esp-now protocol. I know there was a team that started creating something like that but i never got it working for me.

Are you willing to create something like this?  I think it would make esp-now more useful in the future.

poky, 2022-10-05:
I think esp-now is a proprietary protocol, however, if you really want to run it on RPi, you could use ESP8266/ESP32 as a bridge, write some UART to esp-now interfaces to complete the task.

nielsnl68, 2022-10-05:
That is indeed an option to go. but something native is imho nicer to have,

grodansparadis, 2022-10-25:
Take a look at https://github.com/thomasfla/Linux-ESPNOW    But you need a wifi adapter with monitor capabilities so the RPi wifi adapter will not do.

leeebo, 2022-10-25:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

nielsnl68, 2022-10-25:
> Take a look at https://github.com/thomasfla/Linux-ESPNOW But you need a wifi adapter with monitor capabilities so the RPi wifi adapter will not do.

yes, i have been following that project, but never got it working or understand how to get it working.
The reason that exist was the reason for me to ask for a official version.

lhespress, 2023-01-11:
@nielsnl68 I agree with @LJYSP , so it's better to following that project.

lhespress, 2023-06-02:
@nielsnl68 Closing this issue since there has been no update on this. Please feel free to reopen if required.

closed 2023-06-02 by lhespress

mithunbabbira, 2026-05-16:
well i did this and its working great for me https://github.com/mithunbabbira/esp_now_framework_uart