espressif/esp-now #51: Callback on send set but never gets called?

Bazmundi, 2022-12-10:
I have two devices, a Wemos BT with battery (aka wemosbat in PlatformIO), and a Wemos D1 R32, with the wemosbat to send a message to the R32 whenever a switch is closed on an input.  The wemosbat is the "Pedal" (switch) and the D1 R32 is the "Pump".  The wemosbat is the AP and the D1 R32 the STA.

Both fire up, connect, at least between the STA and the AP (I've used this scheme plenty of times for MQTT work but not with ESPNow - first time).  However, while I set the callback on send, it never gets called?

Both use:

```
void InitESPNow(void) {
  //If the initialization was successful
  if (esp_now_init() == ESP_OK) {
    Serial.println("\nESPNow Init Success");
  }
  //If there was an error
  else {
    Serial.println("ESPNow Init Failed");
    ESP.restart();
  }
}
```

### PEDAL

The callback on send is simple and aims to set a flag locally if the callback is successful, since that hints to the PEDAL that the PUMP is present.

```
void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {

  Serial.print("Sent to pump"); 

  if (status == ESP_NOW_SEND_SUCCESS) {
    Serial.println( "Delivery Success");
    pumpPresence = PUMP_PRESENT;
  } else {
    Serial.println( "Delivery Fail");
    pumpPresence = PUMP_UNAVAILABLE;
  }
}
```

Setup reports ESPNow initialise and peer attached.

```
IPAddress local_ip(192,168,4,100);  // set IP of AP device
IPAddress gateway(192,168,4,1);
IPAddress subnet(255,255,255,0);

// setup device
void setup() {
  Serial.begin(115200);

  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if(!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    return;
  }
  
  // setup access point
  WiFi.mode(WIFI_AP);
  delay(1500);
  WiFi.softAPConfig(local_ip, gateway, subnet);
  WiFi.softAP(ssid, password /* , channel, hide_SSID, max_connection*/);
  delay(1500);
  Serial.print("\nMAC: ");
  Serial.println(WiFi.macAddress());

  // start ESP-NOW
  InitESPNow();

  esp_now_register_send_cb(OnDataSent);

  // setup pedal input
  bounce.attach( BOUNCE_PIN ,  INPUT );
  bounce.interval(5); // interval in ms
  displayUpdate(bounce.read(),pumpPresence);

  // LED SETUP
  int ledState = HIGH;
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, ledState);

  // add peer
  memcpy(peerInfo.peer_addr, pumpAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
          
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  } else {
    Serial.println("peer added");

  }

}
```

The main loop will halt and report no PUMP.

```
void findPump (void) {
  while (not pumpPresence) {
    pumpMessage.pumpFlag_msg = PUMP_CONNECT;

    esp_err_t result = esp_now_send(pumpAddress, (uint8_t *) &pumpMessage, sizeof(pumpMessage));
   
    if (result == ESP_OK) {
      Serial.println("Find pump success");
      pumpPresence = PUMP_PRESENT;
    }
    else {
      Serial.println("Pump not there");
      pumpPresence = PUMP_UNAVAILABLE;
      delay(2000);
    }
  }
}

// forever do
void loop() {

  findPump();

  ...
}
```

It is iffy currently, having the PUMP PRESENCE set both in callback and `findPump`, but I am still trying to understand where is best to do this.  I am assuming it will be in the callback, but as I am stuck presently, with the callback not firing, this is something to address once the callback is working.

Console reports:

```
MAC: 80:7D:3A:A3:F9:58
ESPNow Init Success
peer added
Pump not there
Pump not there
Pump not there
**...**
```

Bazmundi, 2022-12-11:
Okay, I am have added more decoding of return from 'esp_now_send`, to get a `ESP NOW peer wifi mismatch'.  Is there a boilerplate for setting up an AP/STA for working with ESP-NOW, please?

Bazmundi, 2022-12-11:
Okay, so I tried the following:

From:
```
  // setup access point
  WiFi.mode(WIFI_AP);
```
To:
```
  // setup access point
  WiFi.mode(WIFI_AP_STA);
  WiFi.disconnect();
```

That seems to have fixed the problem.

I'll do some further checking and report back.

Bazmundi, 2022-12-11:
Seems fine.  I just have to sort now how to get the pair to reconnect if, say, the pump gets turned on then off.  It seems to need some more smarts added.  The function descriptions in the resources don't help with understanding the function relational behaviour.

Bazmundi, 2022-12-12:
So, I am finding that people are either:
- using STA with no WiFi.disconnect
- using STA with WiFi.disconnect on "master"
- using STA on "master" with AP on "slave"

I got away with AP_STA and WiFi.disconnect() on "master".

Is anyone right?  Are they all right? Are any of them wrong?

lhespress, 2023-05-04:
@Bazmundi Any update about this?

Bazmundi, 2023-05-07:
[saga resolved here](https://organicmonkeymotion.wordpress.com/2022/12/12/whos-right/)

Basically, looking at the idea that the framework uses lower end of OSI stack, it makes sense that it is, for both master/slave, to use:

```
WiFi.mode(WIFI_STA);
WiFi.disconnet();
```

I note, however, the odd occasional message that does not get through.  Infrequent, and the error reports seem to be link up, but message not received, but I suspect that is simply par for the course, and some interference.   For my application, the problem is solved by pressing the button a second time, no harm done.

lhespress, 2023-05-08:
@Bazmundi Good news, please close the issue if it has solved.