espressif/esp-now #55: Compilation error if SPIRAM is enabled in master branch

Mr-Techtron, 2023-02-08:
If I enable the SPIRAM in the configuration, there is the compilation error reporting that "CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM" is undeclared, since the WiFi configuration use the static tx buffer. I am using release/v4.4 of ESP-IDF.

wujiangang, 2023-02-08:
@Mr-Techtron Thanks for your report. Please update line https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L51 with
```c
#if CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM
#define MAX_BUFFERED_NUM              (CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM / 2)     /* Not more than CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM */
#elif CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM
#define MAX_BUFFERED_NUM              (CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM / 2)     /* Not more than CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM */
#endif
```
A fix will be soon in the master branch.

Mr-Techtron, 2023-02-08:
@wujiangang Thank you. I have already done that.

closed 2023-02-08 by Mr-Techtron