espressif/esp-now #56: Versioning mess and build errors on IDF 5.0

ramiws, 2023-03-11:
I have been trying for 2 days to get espnow to work at the same when wifi is connected. I have IDF 5.0 version that comes with internal ESPNOW version 1.0 
I want to upgrade to ESPNOW 2.0  is this repo for that? I assume yes, so I went to component registry and installed the ESPNOW component 2.0. It says anything higher than IDF 4.3 is supported ... 

`
/Users/,,,,/,,,,/..../managed_components/espressif__esp-now/src/debug/src/commands/cmd_wifi.c:322:67: error: implicit declaration of function 'ip_2_ip6'; did you mean 'ip_2_ip4'? [-Werror=implicit-function-declaration]
  322 |         ESP_LOGI(TAG, "\n--- %s ping statistics ---", inet6_ntoa(*ip_2_ip6(&target_addr)));
`

Then this repo throw you to the docs of IDF which is only about ESPNOW 1.0 
I am very frustrated, my whole experience with IDF is just very bad docs, broken links.. etc

lhespress, 2023-03-13:
@ramiws Please enable `CONFIG_LWIP_IPV6` config.

wujiangang, 2023-03-13:
@ramiws Sorry that it makes confusing to you. Here I would give some clarification, the ESP-NOW version 1.0 provided by ESP-IDF is the version of ESP-NOW protocol self, in order to simplify the use of ESP-NOW and provide some high-level functionalities, this repository/component provides application-level support for ESP-NOW, including [these features](https://github.com/espressif/esp-now/blob/master/User_Guide.md#function), all these features are based on ESP-NOW protocol 1.0, the version of this component is not equal to the version of ESP-NOW protocol. We will update the document to make it more clear. For the compile issue, we will add some checks when IPv6 is disabled.
Thank you for the helpful information.

ramiws, 2023-03-13:
Okay, yes I reached the same conclusion after studying the 2.0 code. Making it clear in docs will save people time.

closed 2023-03-13 by ramiws