espressif/esp-now #60: Unable to get security sampe to work (idf 5.0.1)

grodansparadis, 2023-03-16:
Can't get the security sample to work. I am on two boards with some ESP32's and idf 5.0.1   Output as follows from standard examples

Any hint to sole appreciated  

Initiator
----------

entry 0x40080638
I (27) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (27) boot: compile time 17:12:34
I (27) boot: chip revision: v1.0
I (30) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (44) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e44ch (123980) map
I (134) esp_image: segment 1: paddr=0002e474 vaddr=3ffb0000 size=01ba4h (  7076) load
I (137) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=6ea10h (453136) map
I (264) esp_image: segment 3: paddr=0009ea38 vaddr=3ffb1ba4 size=00ce0h (  3296) load
I (266) esp_image: segment 4: paddr=0009f720 vaddr=40080000 size=151cch ( 86476) load
I (309) boot: Loaded app from partition at offset 0x10000
I (309) boot: Disabling RNG early entropy source...
I (320) cpu_start: Pro cpu up.
I (320) cpu_start: Single core mode
I (328) cpu_start: Pro cpu start user code
I (328) cpu_start: cpu freq: 160000000 Hz
I (329) cpu_start: Application information:
I (334) cpu_start: Project name:     security_initiator
I (339) cpu_start: App version:      02da898-dirty
I (345) cpu_start: Compile time:     Mar 16 2023 17:12:27
I (351) cpu_start: ELF file SHA256:  d8a204f8e01ed2b9...
I (357) cpu_start: ESP-IDF:          v5.0.1
I (362) cpu_start: Min chip rev:     v0.0
I (367) cpu_start: Max chip rev:     v3.99 
I (371) cpu_start: Chip rev:         v1.0
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (389) heap_init: At 3FFB5CF0 len 0002A310 (168 KiB): DRAM
I (396) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (402) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (408) heap_init: At 400951CC len 0000AE34 (43 KiB): IRAM
I (414) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (421) spi_flash: detected chip: gd
I (425) spi_flash: flash io: qio
I (429) cpu_start: Starting scheduler on PRO CPU.
I (448) wifi:wifi driver task: 3ffbc26c, prio:23, stack:6656, core=0
I (448) system_api: Base MAC address is not set
I (452) system_api: read default base MAC address from EFUSE
I (460) wifi:wifi firmware version: 17afb16
I (462) wifi:wifi certification version: v7.0
I (466) wifi:config NVS flash: enabled
I (470) wifi:config nano formating: disabled
I (474) wifi:Init data frame dynamic rx buffer num: 32
I (478) wifi:Init management frame dynamic rx buffer num: 32
I (484) wifi:Init management short buffer num: 32
I (488) wifi:Init dynamic tx buffer num: 32
I (492) wifi:Init static rx buffer size: 1600
I (496) wifi:Init static rx buffer num: 10
I (500) wifi:Init dynamic rx buffer num: 32
I (504) wifi_init: rx ba win: 6
I (508) wifi_init: tcpip mbox: 32
I (512) wifi_init: udp mbox: 6
I (515) wifi_init: tcp mbox: 6
[... 459 lines cut ...]
D (1792) espnow_sec_init: ESPNOW_SEC_TYPE_INFO
D (1797) espnow_sec_init: Device information:
D (1801) espnow_sec_init: Device channel:   1
D (1806) espnow_sec_init: Device rssi:      -66
D (1810) espnow_sec_init: Device mac:       30:ae:a4:1e:8d:6c
D (1816) espnow_sec_init: Security information:
D (1821) espnow_sec_init: Version:          0
D (1825) espnow_sec_init: Client MAC:       00:00:00:00:00:00
D (1927) espnow: [espnow_send, 551] addr: ff:ff:ff:ff:ff:ff, size: 1, count: 10, rssi: -65, data: , magic: 0xd1b8
D (2428) espnow: [espnow_send, 551] addr: ff:ff:ff:ff:ff:ff, size: 1, count: 10, rssi: -65, data: , magic: 0xd1b8
D (2929) espnow: [espnow_send, 551] addr: ff:ff:ff:ff:ff:ff, size: 1, count: 10, rssi: -65, data: , magic: 0xd1b8
D (3431) espnow: [espnow_send, 551] addr: ff:ff:ff:ff:ff:ff, size: 1, count: 10, rssi: -65, data: , magic: 0xd1b8
W (3933) app_main: espnow wait security num: 1
I (3933) espnow_sec_init: count: 0, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
D (3939) client_security1: Start to write setup0_command
D (4116) client_security1: Write setup0_command done
D (4117) espnow: [espnow_send, 551] addr: 53:45:43:00:00:00, size: 43, count: 1, rssi: -65, data: )Z%�"
 w�i0�f��@}1��r�!�P#��1��լS, magic: 0xd3e3
I (5627) espnow_sec_init: count: 1, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
D (5629) client_security1: Start to write setup0_command
D (5807) client_security1: Write setup0_command done
D (5808) espnow: [espnow_send, 551] addr: 53:45:43:00:00:00, size: 43, count: 1, rssi: -65, data: )Z%�"
 �����#|OFo�n;{��4G�<��CW�L��٣s?�, magic: 0x53ec
I (7318) app_main: App key is sent to the device to complete, Spend time: 5903ms, Scan time: 2518ms
I (7318) app_main: Devices security completed, successed_num: 0, unfinished_num: 1

grodansparadis, 2023-03-16:
Tested also on 5.0.0. Same behavior

responser

I (542) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (627)I (628) espnow: esp-now Version: 2.1.0
I (629) app_main: Uart read handle task is running
I (629) ESPNOW: espnow [version: 1.0] init
I (631) espnow: mac: cc:50:e3:80:0e:28, version: 2
I (636) espnow: Enable main task
I (641) espnow: main task entry

Initiator

I (542) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (63��I (634) espnow: esp-now Version: 2.1.0
I (635) app_main: Uart read handle task is running
I (635) ESPNOW: espnow [version: 1.0] init
I (638) espnow: mac: cc:50:e3:80:10:bc, version: 2
I (643) espnow: Enable main task
I (648) espnow: main task entry
W (3155) app_main: espnow wait security num: 1
I (3155) espnow_sec_init: count: 0, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
I (4818) espnow_sec_init: count: 1, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
I (6473) app_main: App key is sent to the device to complete, Spend time: 5824ms, Scan time: 2506ms
I (6473) app_main: Devices security completed, successed_num: 0, unfinished_num: 1

lhespress, 2023-03-17:
@grodansparadis Please add the patch as the attachment and check again:
[init.zip](https://github.com/espressif/esp-now/files/11000072/init.zip)

Initiator
```
I (545) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (639) wI (640) espnow: esp-now Version: 2.1.0
I (641) app_main: Uart read handle task is running
I (642) ESPNOW: espnow [version: 1.0] init
I (644) espnow: mac: c0:49:ef:bd:4e:9c, version: 2
I (649) espnow: Enable main task
I (654) espnow: main task entry
W (3161) app_main: espnow wait security num: 1
I (3161) espnow_sec_init: count: 0, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
I (3925) app_main: App key is sent to the device to complete, Spend time: 3272ms, Scan time: 2508ms
I (3926) app_main: Devices security completed, successed_num: 1, unfinished_num: 0
```
Responder

```
I (544) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (550) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1059) I (1060) espnow: esp-now Version: 2.1.0
I (1061) app_main: Uart read handle task is running
I (1062) ESPNOW: espnow [version: 1.0] init
I (1064) espnow: mac: c8:f0:9e:35:93:70, version: 2
I (1069) espnow: Enable main task
I (1073) espnow: main task entry
I (8855) espnow_sec_resp: Get APP key
```

grodansparadis, 2023-03-17:
Thanks for a quick response patch that starts the show again.

closed 2023-03-17 by grodansparadis