espressif/esp-now #61: bug in espnow_prov_responder_start

grodansparadis, 2023-03-22:
As I understand, in the way [espnow_prov_responder_start](https://github.com/espressif/esp-now/blob/master/src/provisioning/src/espnow_prov.c#L266)  is written now it can't handle user data. It currently just use the size of **espnow_prov_responder_t** but should use size of the structure + custom_size.  

Maybe __attribute__((packed)) would be a good idea to.

IMHO a better approach would be to copy in a void data with a specific size and interpret what data it is in user space. This give a lot more freedom to use this for other purposes then just wifi provisioning.

lhespress, 2023-03-23:
@grodansparadis What do you mean need include user data by `espnow_prov_responder_t`? just like this:

```
typedef struct {
    char product_id[16];
    char device_name[16];
    void *user_data;
    uint16_t user_size;
} espnow_prov_responder_t;
```

grodansparadis, 2023-03-23:
But the user data is never sent because user_size is never taken into account.

lhespress, 2023-03-23:
@grodansparadis If the structure is like this. It need copy the data to the structure when the user_size is not zero. I just want to confirm is it match your requirement.

grodansparadis, 2023-03-23:
Yes I understand. But the sent size should IMHO be sizeof(espnow_prov_responder_t) + user_size Not like in

https://github.com/espressif/esp-now/blob/master/src/provisioning/src/espnow_prov.c#L275 

or do I miss something?

The schema is otherwise perfect and usable.  In my case I need to send some other info than the wifi info so I would have preferred a void data pointer + a size for the hole thing. The receiver must know anyway what to expect. But this works fine as well.

grodansparadis, 2023-03-23:
@lhespress  Sorry for my last confusing response :/ . I  tested to use the user data in wifi_config_t and that is what I refered to.

But absolutly

typedef struct {
    char product_id[16];
    char device_name[16];
    void *user_data;
    uint16_t user_size;
} espnow_prov_responder_t;

would be perfect.

grodansparadis, 2023-03-23:
Just a note. Other parts of esp-now use a construct like

data size
user_data[]

so a struct like may be best

typedef struct {
char product_id[16];
char device_name[16];
uint8_t user_size;
uint8_t user_data[0];
} espnow_prov_responder_t;

so to be consistent with espnow_prov_wifi_t and others.   Not a big deal of course.

lhespress, 2023-03-23:
@grodansparadis Aha, i have noticed that, thanks again.

closed 2023-04-12 by grodansparadis