espressif/esp-now #62: Problem when init security key from wifi enabled node to espnow node

grodansparadis, 2023-03-28:
I still have some problems with transfer of the security key to a node that need it. My plan has been to have an init button on all nodes and espnow nodes get there security keys from wifi enabled nodes which are initiators. That way forming a cluster of nodes around a  wifi enabled node.

So when I press the init button on a responder node it executes [espnow_sec_responder_start(POP);](https://github.com/espressif/esp-now/blob/master/src/security/src/espnow_security_responder.c#L201) and on the initiator when I press the button it executes [security sample](https://github.com/espressif/esp-now/blob/master/examples/security/main/app_main.c#L135).  

This works perfect if the initiator nodes wifi is not provisioned. But as soon as it is it does not work. It appears that the espnow node does not get any messages from the initiator. 

I must have missed something and don't have a clue right now what to add/change.   Any hints are welcome.

grodansparadis, 2023-03-29:
To add to this.  If I manually set the same channel on the responder as the initiator everything works as I expect.  The question is still how to let it know the channel without wifi credentials.

lhespress, 2023-03-30:
@grodansparadis The root reason is that the responder node doesn't know the initiator's channel when it connected with wifi and change the channel. So you need add a method to sync the channel:

1. Responder announce  a `SYNC` message on all channels periodicity which only initiator response the `ACK`, and then set the channel after receive the `ACK` from the initiator.
2. Store the channel firstly when initiator connect with the wifi, then wait for disconnected and announce  a `SYNC` message on all channels which  only responder response the `ACK`. connect the wifi again after receive the `ACK` from the responder. Responder set the `SYNC` channel after response the 'ACK'.

grodansparadis, 2023-03-30:
@lhespress  Hi thanks for your always kind and prompt response. To move along with my project I added some probe  code yesterday that probe all channels for a counterpart and find the channel and that solve the problem. I just thought this was in the espnow  code somewhere and that I misses something.  Thanks for clarifying!

closed 2023-03-30 by grodansparadis