espressif/esp-now #64: Can one espnow work as Initiator and responder at same  time?

horsemann07, 2023-03-31:
Hi all, 
For some project, I want to one esp32 which will be master & should work as initiator and responder at same time as per required condition? So,  can we achieve that with this repository?

If yes, can you give some info how it can be?

Thanks!

lhespress, 2023-03-31:
@horsemann07 The [control example](https://github.com/espressif/esp-now/tree/master/examples/control) can work on both Initiator and responder.

horsemann07, 2023-03-31:
Hello @lhespress, I did not get that, So, you are saying control example works as both initiator and responder at same time?

My requirement is -> Consider 3 ESP device A (master), B & C as slave. Now A initiated security process and with B and started send/receive data. Now at same time C want to send data to A. So, C initiated security process with A, so, now A should work as security responder at same time it should also keep connection with B.

I check the source code, I think at one time only secure connection can made. If I'm wrong, please correct me.

lhespress, 2023-03-31:
@horsemann07 Yes, it show the clear data how to work on initiator and responder. For your requirement, the device A can security handshake with B and C at same time. So you don't need the second handshake. Then they can work just as control examples.

horsemann07, 2023-03-31:
okay, thanks for the information. I will check that.

closed 2023-03-31 by horsemann07