espressif/esp-now #65: Order of fields in ESPNOW_INIT_CONFIG_DEFAULT doesn't match declaration

vortex314, 2023-04-02:
Got compiler errors : 
```
note: in expansion of macro 'ESPNOW_INIT_CONFIG_DEFAULT'
   37 |         espnow_config_t espnow_config = ESPNOW_INIT_CONFIG_DEFAULT();
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/lieven/workspace/esp-now-bridge/managed_components/espressif__esp-now/src/espnow/include/espnow.h:126:5: error: designator order for field 'espnow_config_t::qsize' does not match declaration order in 'espnow_config_t'
```
Changing the order correctly in the below include works. 
```
typedef struct {
    const uint8_t pmk[16];                  /**< Primary master key */
    bool forward_enable             : 1;    /**< Forward when packets are received */
    bool forward_switch_channel     : 1;    /**< Forward data packet with exchange channel */
    bool sec_enable                 : 1;    /**< Encrypt ESP-NOW data payload when send and decrypt when receive */
    uint8_t reverse                 : 5;
    uint8_t qsize;                          /**< Size of packet buffer queue */
    uint8_t send_retry_num;                 /**< Number of retransmissions */
    uint32_t send_max_timeout;              /**< Maximum timeout */
    struct {
        bool ack                    : 1;
        bool forward                : 1;
        bool group                  : 1;
        bool provisoning            : 1;
        bool control_bind           : 1;
        bool control_data           : 1;
        bool ota_status             : 1;
        bool ota_data               : 1;
        bool debug_log              : 1;
        bool debug_command          : 1;
        bool data                   : 1;
        bool sec_status             : 1;
        bool sec                    : 1;
        bool sec_data               : 1;
        uint32_t reserved           : 18;
    } receive_enable;            /**< Set 1 to enable receiving the corresponding ESP-NOW data type */
} espnow_config_t;

#define ESPNOW_INIT_CONFIG_DEFAULT() { \
    .pmk = "ESP_NOW", \
    .forward_enable = 1, \
    .forward_switch_channel = 0, \
    .sec_enable = 0, \
    .send_retry_num = 10, \
    .send_max_timeout = pdMS_TO_TICKS(3000),\
    .qsize = 32, \
    .receive_enable = { \
                .ack           = 1, \
                .forward       = 1, \
                .group         = 1, \
                .provisoning   = 0, \
                .control_bind  = 0, \
                .control_data  = 0, \
                .ota_status    = 0, \
                .ota_data      = 0, \
                .debug_log     = 0, \
                .debug_command = 0, \
                .data          = 0, \
                .sec_status    = 0, \
                .sec           = 0, \
                .sec_data      = 0, \
                .reserved      = 0, \
                }, \
    }
```

lhespress, 2023-04-03:
@vortex314 Can you tell me how to show the issue? I added it via `idf.py add-dependency "espressif/esp-now^2.1.1"` based on the espnow example in 5.0.0 and 5.0.1, and modify the code as the attachment, it's fine.
[espnow_compile.zip](https://github.com/espressif/esp-now/files/11134545/espnow_compile.zip)

vortex314, 2023-04-03:
I copied the component from managed components to my own components and changed the order of fields in :  I moved 'qsize=32' , 2 lines earlier and then it compiles. 
But I notice also that the esp_now api is quiet different between the main program and the component I installed. Not sure what version mixing has been happening. I'll have to retry from scratch.

vortex314, 2023-04-03:
Maybe the difference is that I compile from C++ and the example is C based. Has ESP-NOW been tested with C++ ? 
Maybe C++ is more order sensitive than C in that sense.

vortex314, 2023-04-03:
Looks highly related to this : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html#designated-initializers

lhespress, 2023-04-03:
@vortex314 It doesn't test with C++, could you provide your example code?

vortex314, 2023-04-04:
@lhespress you can find the source here : https://github.com/vortex314/esp-now-bridge/blob/main/main/main.cpp
I solved it by creating myself the ESPNOW_INIT_CONFIG_DEFAULT initialization structure in a C++ compliant way.

lhespress, 2023-04-10:
@vortex314 Thanks for your feedback, i have reproduce the issue as your source.

closed 2023-04-10 by zhanzhaocheng