espressif/esp-now #67: espnow_security examples not working

horsemann07, 2023-04-07:
Hello,

I was trying to run the espnow security example, I connected two esp32, one responder and one initiator. 
On initiator side, I did not receive any responder info. I tried to increase the time of initiator stills result is same.
Before I tried esp-now release v 1.0.0 there it was working properly. But in esp-now v2.1.0 I m facing issue.

Env:
- ESP-IDF v4.4 IDF version used
- ESP32
- Linux Env

Initiator Log:
```
I (538) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (630)I (631) espnow: esp-now Version: 2.1.1
I (632) app_main: Uart read handle task is running
I (632) ESPNOW: espnow [version: 1.0] init
I (634) espnow: mac: 58:bf:25:99:ab:08, version: 2
I (639) espnow: Enable main task
I (644) espnow: main task entry
W (3151) app_main: espnow wait security num: 0
```
Respomder Log:

```
I (529) wifi_init: WiFi IRAM OP enabled
I (534) wifi_init: WiFi RX IRAM OP enabled
I (539) wifi:Set ps type: 0

I (542) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (631)I (633) espnow: esp-now Version: 2.1.1
I (633) app_main: Uart read handle task is running
I (634) ESPNOW: espnow [version: 1.0] init
I (636) espnow: mac: 58:bf:25:36:7d:2c, version: 2
I (641) espnow: Enable main task
I (646) espnow: main task entry

```
Did anyone test the security example?

lhespress, 2023-04-10:
@horsemann07 I run the command ```idf.py create-project-from-example "espressif/esp-now^2.1.1:security"``` and test it on ESP32 use [IDF v4.4.4](https://github.com/espressif/esp-idf/tree/v4.4.4), it works fine:

Initiator Log:
```
I (539) wifi:Set ps type: 0

I (542) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (639)I (640) espnow: esp-now Version: 2.1.1
I (641) app_main: Uart read handle task is running
I (641) ESPNOW: espnow [version: 1.0] init
I (643) espnow: mac: c0:49:ef:bd:4e:9c, version: 2
I (648) espnow: Enable main task
I (653) espnow: main task entry
W (3160) app_main: espnow wait security num: 1
I (3160) espnow_sec_init: count: 0, Secure_initiator_send, requested_num: 1, unfinished_num: 1, successed_num: 0
I (4274) app_main: App key is sent to the device to complete, Spend time: 3620ms, Scan time: 2506ms
I (4275) app_main: Devices security completed, successed_num: 1, unfinished_num: 0
```

Respomder Log:

```
I (528) wifi_init: WiFi IRAM OP enabled
I (533) wifi_init: WiFi RX IRAM OP enabled
I (538) wifi:Set ps type: 0

I (541) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (547) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (872)I (873) espnow: esp-now Version: 2.1.1
I (874) app_main: Uart read handle task is running
I (874) ESPNOW: espnow [version: 1.0] init
I (876) espnow: mac: c8:f0:9e:35:93:70, version: 2
I (881) espnow: Enable main task
I (886) espnow: main task entry
I (13884) espnow_sec_resp: Get APP key
```

can you check it again.

horsemann07, 2023-04-10:
Hi @lhespress 

Thanks for the verifying on your side, Its working now. I have to clone the repository  again to work. Not sure why before it was not working, 

Thanks for the reply!

horsemann07, 2023-04-10:
Now I understand why it not working previously. I made some changes to the espnow code [here](https://github.com/espressif/esp-now/blob/ad532ef73085507f55980015f59c9021ca0d8023/src/espnow/src/espnow.c#L79). 

There I remove other in tnings in  `espnow_frame_format_t` stucture and except `vendor_specific_content`. 

I m little confuse on this, `espnow_frame_format_t` is given as example for user or it used by espnow lower layer, so any modification in that structure leads to problem?

lhespress, 2023-04-13:
@horsemann07 This project care some filtration information, eg: [RSSI](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L210). But, it doesn't have API or parameter to get [wifi_pkt_rx_ctrl_t](https://github.com/espressif/esp-now/blob/master/src/espnow/src/espnow.c#L182) from receiving data before [ESP-IDF v5.0.1](https://github.com/espressif/esp-idf/blob/v5.0.1/components/esp_wifi/include/esp_now.h#L90), so given 
`espnow_frame_format_t` which be match with [ESP-NOW Frame Format](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html) as an example for user get `wifi_pkt_rx_ctrl_t`, just as you point, any modification in that structure will lead problem.
The user doesn't care this as `esp_now_recv_info_t` include it from [ESP-IDF v5.0.1](https://github.com/espressif/esp-idf/blob/v5.0.1/components/esp_wifi/include/esp_now.h#L90).

horsemann07, 2023-04-13:
Thanks @lhespress for clearing my confusion.

closed 2023-04-13 by horsemann07