espressif/esp-now #68: ESPNOW fails because of ESP_ERR_ESPNOW_IF

lonelysheep1212, 2023-04-08:
I want to scan the RSSI value of the surrounding device after using wifi, and then send it to the corresponding device through ESPnow, but the program keeps reporting an error.
The programs are as follows：
wifi_init：
[image]
wifi_scan:
[image]
When wifi_scan has done, esp32 got the information,and runs esp_wifi_stop();
wifi init agin:
[image]
esp_now init :
[image]
send data:
[image]
The results are as followed:
[image]
[image]

I don't have a clue about this error, thank you very much if you can help answer it

lhespress, 2023-04-10:
@lonelysheep1212 It's not necessary runs `esp_wifi_stop()` and `esp_wifi_start()` after got the RSSI info. The error means that current WiFi interface doesn't match that of peer, please check the log as follows before `esp_now_init`:

```
I (3607) wifi:flush txq
I (3607) wifi:stop sw txq
I (3607) wifi:lmac stop hw txq
I (3617) wifi:mode : sta (c8:f0:9e:35:93:70)
I (3617) wifi:enable tsf
I (3617) ESPNOW: espnow [version: 1.0] init
```

lonelysheep1212, 2023-04-10:
> @lonelysheep1212 It's not necessary runs `esp_wifi_stop()` and `esp_wifi_start()` after got the RSSI info. The error means that current WiFi interface doesn't match that of peer, please check the [...]
Thanks for your feedback, i'll check it again.

lonelysheep1212, 2023-04-10:
> 没有必要运行，并且在获得 RSSI 信息后。该错误表示当前WiFi接口与对等方的WiFi接口不匹配，请按以下方式检查日志：`esp_wifi_stop()``esp_wifi_start()``esp_now_init`
> 
> ```
> I (3607) wifi:flush txq
> I (3607) wifi:stop sw txq
> I (3607) wifi:lmac stop hw txq
> I (3617) wifi:mode : sta (c8:f0:9e:35:93:70)
> I (3617) wifi:enable tsf
> I (3617) ESPNOW: espnow [version: 1.0] init
> ```

Thanks for your help， I have solved  problem。

closed 2023-04-10 by lonelysheep1212