espressif/esp-now #69: Disable CCA(Carrise sense) using ESP-NOW ???

uecken, 2023-04-12:
Does anyone know disable the CCA(Carrise sense) using ESP-NOW?
I think below function means the carrier sense threshold.
esp_ieee802154_set_cca_threshold(int8_t cca_threshold);

Therefore, esp_ieee802154_set_cca_threshold(-200) disable CCA?

https://github.com/espressif/arduino-esp32/blob/00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff/tools/sdk/esp32c3/include/ieee802154/include/esp_ieee802154.h

IDE: PlatformI
HW: ESP32-C3

lhespress, 2023-04-13:
@uecken Please add the patch to the examples: [get_start.diff](https://github.com/espressif/esp-now/tree/master/examples/get-started) and [cca.patch](https://github.com/espressif/esp-idf/tree/v5.0.1/examples/wifi/espnow) as the attachment [cca.zip](https://github.com/espressif/esp-now/files/11217621/cca.zip)

lhespress, 2023-06-02:
@uecken Closing this issue since there has been no update on this. Please feel free to reopen if required.

closed 2023-06-02 by lhespress

uecken, 2023-06-02:
Now, I noticed your reply. Thank you.

uecken, 2023-06-02:
Sorry, my information was lost. I means arduino environment.
How about below envronment?

[env:esp32-c3-devkitm-1]
platform = espressif32
board = seeed_xiao_esp32c3
framework = arduino