espressif/esp-now #71: Encrypted logging

grodansparadis, 2023-04-18:
I can solve this in other ways but it would have been nice if espnow remote logging could be encrypted as well. This due to sensitive information that often is shared in the logs. If remote log frames are encrypted one can safely diagnose a remote device by enabling logging on it in a live environment. 

It is only a change in  espnow_log_send_task at https://github.com/espressif/esp-now/blob/a90570c9be583f2b9b2dfc102a7736d90c64e206/src/debug/src/espnow_log.c#L81

that is needed using the espnow_frame_head_t as in other places of the code and add a "secure flag" in the log config. If this switch is false it should not break existing code. 

But as I sad, this is more of a feature request and I can solve this by other means. I have probably reached my feature requests limit by now ... :)

lhespress, 2023-04-19:
@grodansparadis Good idea!! We'll think about whether all components support it by default.

lhespress, 2023-04-21:
@grodansparadis Please add the patch as the attachment which support security for all components.

grodansparadis, 2023-04-21:
@lhespress It's a good friday. :)  Now have a very good weekend.

grodansparadis, 2023-04-24:
@lhespress For me this patch breaks the key exchange, the security initiator does no longer find any clients expecting keys. I am investigating this so I have't introduced something in my code that make this fail but are a bit delayed in this by other things so I think I mention it here in case you see something obvious

grodansparadis, 2023-04-24:
@lhespress If I understand this correctly both "protocomm_espnow_initiator_start" and "espnow_sec_initiator_scan" should have a frame_header with .security=false  But I am tired now and it's late so I may very well be dead wrong... ;-)

lhespress, 2023-04-25:
@grodansparadis Yeah, your understand is correct, `security` should have a frame_header with .security=false for negotiation keys.

grodansparadis, 2023-04-25:
@lhespress  Sorry for this one. I messed up in my own code (as usual :) ).   The patch works perfect.  Thanks for your good work and kind responses/help as always..

closed 2023-04-25 by grodansparadis

lhespress, 2023-04-25:
@grodansparadis It doesn't matter. BTW, Would you mind sharing some information about your project?

grodansparadis, 2023-04-25:
@lhespress I am the maintainer of an open and free protocol called [VSCP](https://vscp.org) that has been around for many (MANY) years. It is a common way for IoT/m2m nodes to communicate with each other and can be used over different media such as Ethernet/CAN/RS-232/422/485/BLE etc etc  Some quick info is [here](https://en.wikipedia.org/wiki/Very_Simple_Control_Protocol)  describing level I of VSCP.
 
I am working with some code that use VSCP over espnow at the moment as they match very well together.  There is some **initial** info on what I try to do here  https://github.com/grodansparadis/vscp-espnow   Very early stuff.

I will release some open hardware later that demonstrate and use this ([like this module](https://github.com/grodansparadis/can4vscp-frankfurt-wifi)).

Cheers
/Ake

lhespress, 2023-04-26:
@grodansparadis Thanks for your sharing, it's a great job.