espressif/esp-now #76: Problem downloading coin_cell_demo using create-project-from-example command

rglissmann, 2023-06-11:
I am trying to download the coin_cell_demo example using the command: 
idf.py create-project-from-example "espressif/esp-now=*:coin_cell_demo" 
and I get the following error: 
ERROR: Cannot find example "coin_cell_demo" for "espressif/esp-now" version "=*"

I don't have a problem downloading the other esp-now examples using this command.

wujiangang, 2023-06-12:
Hi @rglissmann Thanks for your report, the command should be `idf.py create-project-from-example "espressif/esp-now=*:coin_cell_demo/bulb"` or `idf.py create-project-from-example "espressif/esp-now*:coin_cell_demo/switch"`, since coin_cell_demo is not the actual example. The README will be fixed later.

rglissmann, 2023-06-12:
Thanks for your quick response. One correction, a missing = sign. idf.py create-project-from-example "espressif/esp-now=*:coin_cell_demo/switch"

closed 2023-06-16 by rglissmann