espressif/esp-now #78: espnow version

bhuvanchandra, 2023-06-21:
The current versioning across different platforms appears rather perplexing. GitHub shows the latest release as v1.0, the ESP registry lists the most recent version as 2.3.0, and the ESPNow API reports its version as v1.0 when used without adding ESPNow as an external component dependency in the v5.0.2 ESP-IDF. Could anyone guide me as to which is the most relevant version to keep track of?

lhespress, 2023-06-25:
@bhuvanchandra The master branch on [GitHub](https://github.com/espressif/esp-now) is match with [Component Register](https://components.espressif.com/components/espressif/esp-now) release version. You can use the master branch on GitHub if you don't add it as  an external component.

bhuvanchandra, 2023-06-26:
@lhespress Got it thanks for your suggestion, will do it.

Just a follow-up question, eventually there will be a release or tag in the espnow repo?

n3b0j5a, 2023-07-24:
@bhuvanchandra  You always can use github as a source for component manager:

```
dependencies:
  esp-now:
    version: master
    path: .
    git: https://github.com/espressif/esp-now.git
  idf:
    version: ">=4.1.0"
  ```
  Version can point to a branch, tag or commit hash.

closed 2024-01-25 by lhespress