espressif/esp-now #79: OTA doesn't work with encryption enabled

n3b0j5a, 2023-06-22:
I have two boards, one based on ESP32-S3 as an initiator and another one based on ESP32-C3 as responder and I'm trying to use S3 board to download update for C3 and push it. I customized OTA download process a bit to achieve 2 things:
* Disable firmware verification to prevent failure due to chip variant difference
* Download firmware to an nvs partition instead of ota_x. I want to preserve initiator ota partitions for eventual rollback.

Both chips are development encrypted.  However, when trying to transfer firmware to C3 device, i get the error code:

```
I (38024) espnow_ota_responder: write packet size: 214
E (38024) spi_flash: flash encrypted write length must be multiple of 16
W (38034) espnow_ota_responder: [espnow_ota_write, 283] <ERROR> esp_partition_write ESP_ERR_INVALID_SIZE
W (38044) espnow_ota_responder: [espnow_ota_responder_data_process, 432] <ERROR> espnow_ota_handle
I (38054) espnow_ota_responder: write packet size: 214
E (38054) spi_flash: flash encrypted write address must be 16 bytes aligned
W (38064) espnow_ota_responder: [espnow_ota_write, 283] <ERROR> esp_partition_write ESP_ERR_INVALID_ARG
W (38074) espnow_ota_responder: [espnow_ota_responder_data_process, 432] <ERROR> espnow_ota_handle
I (38084) espnow_ota_responder: write packet size: 214
E (38094) spi_flash: flash encrypted write address must be 16 bytes aligned
W (38094) espnow_ota_responder: [espnow_ota_write, 283] <ERROR> esp_partition_write ESP_ERR_INVALID_ARG
W (38104) espnow_ota_responder: [espnow_ota_responder_data_process, 432] <ERROR> espnow_ota_handle
I (38114) espnow_ota_responder: write packet size: 214
E (38124) spi_flash: flash encrypted write address must be 16 bytes aligned
W (38134) espnow_ota_responder: [espnow_ota_write, 283] <ERROR> esp_partition_write ESP_ERR_INVALID_ARG
W (38144) espnow_ota_responder: [espnow_ota_responder_data_process, 432] <ERROR> espnow_ota_handle
```
I suspect that this is due to the fact that flash must be written in 16 bytes blocks when encryption is enabled.

Demo code listing for initiator:

```
#include "sys/param.h"

#include "esp_log.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/timers.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_mac.h"
#include "esp_flash_encrypt.h"
#include "esp_app_format.h"
#include "esp_image_format.h"
#include "esp_http_client.h"
#include "esp_ota_ops.h"
#include "espnow_utils.h"
#include "espnow.h"
#include "espnow_ota.h"
#include "espnow_ctrl.h"
#include "spi_flash_mmap.h"

#define UNBIND_TOTAL_COUNT               (5)

static const char *TAG = "APP";

static led_strip_handle_t g_strip_handle = NULL;

/* Partial_data is word aligned so no reallocation is necessary for encrypted flash write */
typedef struct {
    const esp_partition_t *part;
    bool need_erase;
    uint32_t wrote_size;
    uint8_t partial_bytes;
    WORD_ALIGNED_ATTR uint8_t partial_data[16];
} ota_ops_entry_t;

esp_err_t app_ota_begin(const esp_partition_t *partition, size_t image_size, ota_ops_entry_t *out_entry)
{
    esp_err_t ret = ESP_OK;

    if (partition == NULL) {
        return ESP_ERR_INVALID_ARG;
[... 381 lines cut ...]

void app_main(void)
{
    app_flash_init();
    espnow_storage_init();
    app_led_init();

    unbind_restore_init();
 
    app_wifi_init();

    espnow_config_t espnow_config = ESPNOW_INIT_CONFIG_DEFAULT();
    espnow_init(&espnow_config);

    esp_event_handler_register(ESP_EVENT_ESPNOW, ESP_EVENT_ANY_ID, app_espnow_event_handler, NULL);

     vTaskDelay(pdMS_TO_TICKS(5000));
    const esp_partition_t *data_partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS, "nvs_ota");
    size_t firmware_size = app_firmware_download(data_partition, "http://10.6.6.7/switch.bin");
    uint8_t sha_256[32] = {0};
    esp_partition_get_sha256(data_partition, sha_256);

    app_firmware_send(firmware_size, sha_256);
}
```

n3b0j5a, 2023-06-22:
This solves SPI flash error.
```
#define ESPNOW_OTA_PACKET_MAX_SIZE             ((ESPNOW_DATA_LEN - 4) - (ESPNOW_DATA_LEN -4) % 16)  /**< Maximum length of a single packet transmitted */
```

 However, process stops after last chunk is received. Still investigating....

lhespress, 2023-06-26:
@n3b0j5a You're right, flash must be written in 16 bytes blocks when encryption is enabled. What's the length of the last chunk data?

n3b0j5a, 2023-06-26:
@lhespress  Encrypted firmware is padded anyway, that should not be a problem. I made patch fix that resolve this issue: https://github.com/espressif/esp-now/pull/80

n3b0j5a, 2023-07-11:
Can someone take a look at pull requests? I really need that fixes merged.

lhespress, 2023-07-13:
@n3b0j5a Thanks for your contribution, the [#80](https://github.com/espressif/esp-now/pull/80) has been merged.

closed 2023-07-14 by n3b0j5a