espressif/esp-now #84: Espnow control bugs

n3b0j5a, 2023-07-26:
@lhespress 
I found two more bugs in control source. First, bindlist in not properly loaded upon startup. It is only loaded if I call `espnow_ctrl_responder_bind` which is what i do not want to do every time, it should be loaded from flash. 
Second one, if i call `espnow_ctrl_responder_data` and register callback, and after that i call `espnow_ctrl_responder_bind` things can get messy because callback functions are inside of bindlist structure. I addressed bot problems in PR #83 by removing callback data from bindlist and making sure that bindlist is loaded from flash before use.

lhespress, 2023-07-31:
@n3b0j5a Thanks for your contribution. I submitted some reviews, PTAL.

n3b0j5a, 2023-08-03:
@lhespress  I see no review there?

lhespress, 2023-08-07:
@n3b0j5a I started a review for ```espnow_load_bindlist``` API.

n3b0j5a, 2023-08-14:
@lhespress For some reason I cannot see it. Actually i don't see any comments on that PR

lhespress, 2023-08-15:
@n3b0j5a sorry, i forget to change the review state, you can see it now.

lhespress, 2024-01-31:
@n3b0j5a I have merged it to the master branch.

closed 2024-01-31 by lhespress