espressif/esp-now #86: Espnow won't compile if ESP_CONSOLE_USB_SERIAL_JTAG is set

n3b0j5a, 2023-08-03:
When CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG is set (and secondary console is disabled) espnow wont compile:

```
/home/user/espnow-test/managed_components/espressif__esp-now/src/debug/src/espnow_console.c:214:26: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  214 |             .baud_rate = CONFIG_ESP_CONSOLE_UART_BAUDRATE,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          CONFIG_ESP_CONSOLE_UART_NUM
```

lhespress, 2023-08-29:
@n3b0j5a `espnow_console.c` works on `ESP_CONSOLE_UART`,i'm not sure whether match your requirement only compile it when enable `ESP_CONSOLE_UART`.

n3b0j5a, 2023-08-30:
I'm using C3 variant which do have USB serial console, and RX0 and TX0 pins are used for other purposes. 
However, I have fix for this ready, i will make PR after you merge the first one.

closed 2023-10-10 by zhanzhaocheng