espressif/esp-now #89: issue using espnow on my esp32 wroom and dht11

Brunno93, 2023-08-28:
I've connected a dht11 do pin 14 and a had the readings, but when I simply add a esp_now.h line in the code it stopped reading the sensor values.

Can anyone help. Pls!

lhespress, 2023-08-29:
@Brunno93 Could you share your code on the [get-started](https://github.com/espressif/esp-now/tree/master/examples/get-started) example

Brunno93, 2023-08-29:
#include <Arduino.h>
#include <WiFi.h>
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <esp_now.h>
 
 const unsigned long eventInterval = 1000;
unsigned long previousTime = 0;

 float Temp;
 float Hum;
 
 #define DHTPIN 14
 #define DHTTYPE    DHT11
 
 DHT dht(DHTPIN, DHTTYPE);

typedef struct message {
  float temperature;
  float humidity;
};
 
struct message myMessage;
 
void onDataReceiver(const uint8_t * mac, const uint8_t *incomingData, int len) {
  Serial.println("Message received.");
  // We don't use mac to verify the sender
  // Let us transform the incomingData into our message structure
 memcpy(&myMessage, incomingData, sizeof(myMessage));
 Serial.println("=== Data ===");
 Serial.print("Mac address: ");
 for (int i = 0; i < 6; i++) {
     Serial.print("0x");
     Serial.print(mac[i], HEX);
     Serial.print(":");
 }
    
 Serial.print("\n\nTemperature: ");
 Serial.println(myMessage.temperature);
 Serial.print("\nHumidity: ");
 Serial.println(myMessage.humidity);
 Serial.println();
}
 
void setup() {
 Serial.begin(115200);
 WiFi.mode(WIFI_STA);
 
 // Get Mac Add
 Serial.print("Mac Address: ");
 Serial.print(WiFi.macAddress());
 Serial.println("ESP32 ESP-Now Broadcast");
 
 // Initializing the ESP-NOW
 if (esp_now_init() != 0) {
   Serial.println("Problem during ESP-NOW init");
   return;
 }
 esp_now_register_recv_cb(onDataReceiver);
}
 
void loop() {

  Hum = dht.readHumidity();
  Temp = dht.readTemperature();
  
  

  unsigned long currentTime = millis();
  if (currentTime - previousTime >= eventInterval) {
    
  Serial.print(F("Humidity: "));
  Serial.print(h);
  Serial.print(F("%  Temperature: "));
  Serial.print(t);
    
    previousTime = currentTime;
  }
}

lhespress, 2023-08-30:
@Brunno93 There is some issues need check from the code:
1. Where call the `loop` function which read Humidity and Temperature from DHT11?
2. `onDataReceiver` function only receive both Humidity and Temperature from the other `esp-now` device, it didn't have read option.
3. Could you change the `pin14` to another pin?
4. Please add the code on my provide example links which i can quickly reproduce it.

Brunno93, 2023-08-30:
Thanks for help
I put the espnow.h at the top and review some lines of code about espasyncwerver.h

closed 2023-08-30 by Brunno93