espressif/esp-now #93: 单播广播解析数据不一致。

M3Captain7, 2023-09-27:
乐鑫公司你好，我使用esp32-s3烧录wifi->espnow代码，在menuconfig中修改为station模式；
使用ESP32-C3同样烧录此代码，在menuconfig中修改为soft AP模式；
当soft AP接受的是广播数据时，可以对我设定的payload很好解析如图ABCDE
[image]
此时我将soft AP（esp32c3）重启，soft AP开始接受单播数据，当soft AP接受的是单播数据时，便会出现解析不完整情况如图只解析AB，接受的长度都同样为我设定的15
[image]
我想知道如何在单播情况成功解析数据ABCDE，以及如何控制单播与广播模式。
还有为什么不是双向接受与发送数据，也就是全双工模式，我在使用过程中一直是半双工模式

lhespress, 2023-09-28:
@M3Captain7 这应该是打印问题，你可以通过十六进制打印确认下数据是否是匹配的；全双工模式请参考 https://github.com/espressif/esp-now/tree/master/examples/solution 示例

M3Captain7, 2023-10-19:
谢谢你的回复

closed 2023-10-19 by M3Captain7